ros/ros_comm #1029: improving logging to support yaml and json configuration files

asmodehn, 2017-04-11:
This is a feature request that would be nice to have, instead of relying on the old fileConfig format.
From : https://docs.python.org/2.7/library/logging.config.html#configuration-file-format

> Note
> The fileConfig() API is older than the dictConfig() API and does not provide functionality to cover certain aspects of logging. For example, you cannot configure Filter objects, which provide for filtering of messages beyond simple integer levels, using fileConfig(). If you need to have instances of Filter in your logging configuration, you will need to use dictConfig(). Note that future enhancements to configuration functionality will be added to dictConfig(), so it’s worth considering transitioning to this newer API when it’s convenient to do so. 

For example, the code could search for python_logging.[conf, yaml, json] and choose the method to load the config based on the file extension.

dirk-thomas, 2017-04-11:
This is likely not an enhancement the maintainers will work on in the future. Please consider providing a PR for this feature. Please take into consideration keep existing functionality working (keeping backwards compatibility) to increase the chance it can be merged.

asmodehn, 2017-04-19:
I started a PR here : https://github.com/ros/ros_comm/pull/1033
Thanks for any feedback, I will keep testing it during the next few days.