ros/ros_comm #1030: roslaunch fails if ROS_LOG_DIR is set and ~/.ros was not yet created

Homan42, 2017-04-12:
Hi,

if ROS_LOG_DIR is set to something other than "\~/.ros", this directory is not created. This causes the subprocess.Popen call in nodeprocess.py:298 to fail if its cwd is set to "\~/.ros". That is the case for example during the automatic creation of rosmaster.

Steps to reproduce:
```
rm -r ~/.ros
export ROS_LOG_DIR=~/log
roslaunch <any launchfile>
```

I used this launchfile to produce the error:
```
<launch>
  <param name="robot_description"
    value="&lt;robot&gt;&lt;/robot&gt;" />

  <node name="robot_state_publisher" pkg="robot_state_publisher"
    type="robot_state_publisher" />
</launch>
```

Output:
```
$ roslaunch test.launch
... logging to /home/*****/log/4695aafa-1f12-11e7-9ad6-64006a5db393/roslaunch-state-ub-ph-23654.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://*********/

SUMMARY
========

PARAMETERS
 * /robot_description: <robot></robot>
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
Roslaunch got a 'No such file or directory' error while attempting to run:

/opt/ros/kinetic/lib/robot_state_publisher/robot_state_publisher __name:=robot_state_publisher __log:=/home/*****/log/4695aafa-1f12-11e7-9ad6-64006a5db393/robot_state_publisher-1.log

Please make sure that all the executables in this command exist and have
executable permission. This is often caused by a bad launch-prefix.
The traceback for the exception was written to the log file
[robot_state_publisher-1] killing on exit

```

dirk-thomas, 2017-04-12:
I don't think it is necessary to update ROS to create it on demand.  The simple solution to this is to make sure the directory you are pointing to in the environment variable exists. Please feel free to create a PR to support this or contribute to the [wiki](http://wiki.ros.org/ROS/EnvironmentVariables) clarifying that the directory needs to exist

Homan42, 2017-04-12:
Hi Dirk,

sorry if I didn't make that clear enough, but the problem is not that the directory in ROS_LOG_DIR can't be found. The problem is that the ROS home directory is only created if ROS_LOG_DIR is not set, but its existence is required by roslaunch in any case, because it is used as working directory for all ROS nodes and the ROS master. So what I am saying is that if you set ROS_LOG_DIR to a custom (existing) directory and ~/.ros does not exist, roslaunch will fail. I will clarify that in the problem description above.

Best,
Homan

dirk-thomas, 2017-04-14:
Thanks for clarifying. The following two PR should address the problem: ros/ros#143 ros/ros_comm#1031

closed 2017-04-15 by Homan42

arotyramel, 2019-03-22:
I had the same issue, when ROS_LOG_DIR was set.
For me the problem has resolved when I also set the ROS_HOME env variable.
e.g.
```
mkdir -p /tmp/roslogs
export ROS_LOG_DIR=/tmp/roslogs
export ROS_HOME=/tmp/roslogs
```