ros/ros_comm #1032: Feature request: rospy.Timer callback with arguments

pylipp, 2017-04-18:
I was wondering why the current implementation of [`rospy.Timer`](https://github.com/ros/ros_comm/blob/lunar-devel/clients/rospy/src/rospy/timer.py#L179) does not support callback arguments. [`rospy.Subscriber`](https://github.com/ros/ros_comm/blob/lunar-devel/clients/rospy/src/rospy/topics.py#L514) supports this. 

I solved this by using native Python `threading.Thread`s with a callback along this: 

    def callback(*args, **kwargs):
        rate = rospy.Rate(1)
        while not rospy.is_shutdown():
            # actual processing routine
            rate.sleep()

I'm willing to submit a PR implementing this, however I'd like to know if issues preventing callback args were encountered during the implementation of `rospy.Timer`.

mikepurvis, 2017-04-18:
You can just pass a closure as the callback, or use something like [functools.partial](https://docs.python.org/2/library/functools.html#functools.partial) to curry in the arguments you want passed. Given how flexible Python is as a language, I don't feel rospy needs to support this natively.

dirk-thomas, 2017-04-18:
I agree with the comment from @mikepurvis. You can use any callable and attach your context to it.

Therefore I will close this as "won't fix".

closed 2017-04-18 by dirk-thomas