ros/ros_comm #1034: xmlrpc++ seems to contain an unlicensed file `base64.h`

de-vri-es, 2017-04-20:
The file `base64.h` distributed with xmlrpc++ seems to be an unlicensed file. It appears to originate from an article at code guru [1], where it does not include any license information. 

The file itself contains an email address that is no longer valid. I got an instant bounce when attempting to send an email to it. I also raised an issue at the xmlrpc++ sourceforge page [2], and I contacted code guru for their view on the matter (although I don't believe their view matters).

Without a license we can't technically redistribute the file. So unless we can find a license for the file it seems it needs to be replaced with a different base64 implementation.

See also [3] for a related discussion on discourse.

[1] http://www.codeguru.com/cpp/cpp/cpp_mfc/article.php/c4095/STL-Compatible-Base64-Encoder.htm
[2] https://sourceforge.net/p/xmlrpcpp/discussion/240494/thread/cec40381/
[3] https://discourse.ros.org/t/ros-gnu-gpl-v1-2-issue/1642/4

dirk-thomas, 2017-04-20:
Just for reference: https://github.com/ros/ros_comm/blob/a8c5f9e95e6823417b67ee0da2fa0a91acd97ef5/utilities/xmlrpcpp/include/xmlrpcpp/base64.h

We either need:
* feedback from the original author (which seems difficult)
* replace the file with an existing library (with a compatible license) or
* reimplement the logic.

de-vri-es, 2017-04-20:
A number of possible alternative libraries:
- `libb64` (public domain, http://libb64.sourceforge.net/ )
- `base64` (MIT license, http://base64.sourceforge.net/ )
- `base_n` (MIT license, https://github.com/azawadzki/base-n/)

`libb64` claims high performance and has both a low-level raw buffer interface and a C++ iostream based interface. It is available as system package on Debian Jessie, Ubuntu 14.04+ and Arch Linux, but not Fedora afaict. It can also be built in-tree if that's easier, of course.

`base64` is not really intended as library, and only works with C `FILE` pointers (or requires a bit of extra glue). It should probably be disregarded.

`base_n` is header-only and looks pretty modern but only has an iterator based interface which *might* be somewhat slow. Then again, maybe the compiler can optimize it very well. Plus, the current `base64.h` is also iterator based, so it probably fits as a replacement easier.

None of these three include a test suite, but xmlrpc++ does seem to test base64 values.

I could run a few simple benchmarks between `libb64` and `base_n` if you think its useful.

dirk-thomas, 2017-04-20:
It would be great if you could provide a pull request to integrate one of the alternative libraries. Or only the necessary header to not introduce additional dependencies which might not be available on all platforms which ROS users are building ROS from source on. A benchmark showing that the performance doesn't suffer due to the change would be awesome.

de-vri-es, 2017-04-21:
I did a few benchmarks for the original, base-n and libb64. The source for the benchmark can be found here together with raw results and the graphs below: https://github.com/delftrobotics-forks/ros_comm/tree/base64-benchmark

The benchmarks were run on a reasonably powerful PC with an Intel i7-4790K and 16GB of RAM. The reported time is the number of microseconds it took for 1000 runs of encoding/decoding the same data (or averaged nanoseconds per single run, which is how the graphs present it). It would be useful if someone can verify the results I got below on a different machine, and review the benchmark code to make sure I didn't mess up.

The benchmark shows that in terms of speed, `libb64` is the clear winner when it comes to processing large input. For smaller input (<100 bytes), the results are less clear. For more details I recommend looking at the `.dat` files here: https://github.com/delftrobotics-forks/ros_comm/tree/base64-benchmark/results (or clone the repository and run the benchmark yourself). The first column of the `.dat` files is the size of the unencoded data, the second time to encode, and the third the time to decode it.

Integrating `libb64` was more of a hassle than integrating `base-n`. The library takes a pointer to an output buffer without a size, so care must be taken that the buffer is always large enough to hold the base64 encoded (or decoded) data including padding and inserted line breaks. If we go for `libb64` I should improve the comments near the relevant sections where that is done.

I also added a unit test to make sure the base64 conversions work, since the included tests don't appear to be unit tests. I kinda feel like removing or at least renaming the non-automated tests and using the `test` directory for unit tests. But that can wait.

Encoding in debug mode:
[image]

Encoding in release mode:
[image]

Decoding in debug mode:
[image]

Decoding in release mode:
[image]

wjwwood, 2017-04-22:
Thanks for digging into this and running the benchmarks @de-vri-es! After only reading what you've written here, it seems to me that building `libb64` in source is a good option. You mentioned that it was slightly harder to integrate, do you think it's a barrier to using it? It seems you were able to overcome that in order to do some benchmarks anyways.

Can I take away from this that you think `libb64` is the best solution (regardless of how we integrate it?)

I don't see a `libb64` in Homebrew, but `base64` is there, `base-n` is also missing. That's not a deal breaker, but it would indicate to me that integrating the project into our source tree might be preferable to making it a dependency.

de-vri-es, 2017-04-22:
> Thanks for digging into this and running the benchmarks @de-vri-es! After only reading what you've written here, it seems to me that building libb64 in source is a good option. You mentioned that it [...]

Yes, I suppose based on my results `libb64` is the best option. I really like the simplicity of `base-n`, but it performs worse than the original in release mode. Do we want someone to verify the results before continuing to integrate `libb64`?

Somewhat on a side-note, since `libb64` is public domain it is easier to redistribute than `basen`, though I believe the MIT license shouldn't be a problem either. We would just have to make sure to ship all the relevant licenses with source and binary distributions afaik.

de-vri-es, 2017-11-12:
Since #1046 is merged, this can be closed.

closed 2017-11-12 by de-vri-es