ros/ros_comm #1036: Logging to syslog ?

asmodehn, 2017-04-24:
I couldn't find any way to log into syslog from ROS log (C++ or python).
Neither in code nor in the documentation...

I was wondering what were the different steps needed, in order to implement such a feature, in a way that doesn't change the current behaviour, but just provide the additional functionality to any user who might need it ( when moving from developing a ROS system in a lab, to releasing it onto production on a robot running far away...)

I might be able to do the effort of implementation, but I'd likely need some guidance...
Thanks for any pointers !

dirk-thomas, 2017-04-28:
The `rosconsole` package has different backends (see https://github.com/ros/ros_comm/tree/lunar-devel/tools/rosconsole/src/rosconsole/impl). The backend can only be selected at compile time though (https://github.com/ros/ros_comm/blob/a0f47897e4a5b5bc35ddb59a8637973e60f6bf23/tools/rosconsole/CMakeLists.txt#L13). While that might be enough for your use case it is not very convenient.

asmodehn, 2017-05-02:
Thanks for the pointer for this.

As far as I understand now, to implement a flexible logging system that can log to screen, file, or syslog, while being easily configurable, to decide what logger goes where, I ll need : 

- C++ : log4cxx + using named loggers + extra configuration to add syslog appender(s) to meaningful loggers
- Python : logging + using named loggers + modified configuration to add syslog handler(s) to meaningful loggers

Correct ?

I realize some of these are still in development, but I ll do what I can to keep the ball rolling on logging features, until we get there...