ros/ros_comm #1038: Can rosbag filter handle `tf2_msgs/TFMessage`

artivis, 2017-04-25:
rosbag filter crashes while filtering over a rosbag that as a `tf2_msgs/TFMessage` message, independently from the predicate for that matter : 

`rosbag filter my_bag_with_tf2.bag out.bag "True"`.  

`Traceback (most recent call last):
  File "/tmp/robag/devel/bin/rosbag", line 6, in <module>
    exec(fh.read())
  File "<string>", line 35, in <module>
  File "/tmp/robag/src/ros_comm/tools/rosbag/src/rosbag/rosbag_main.py", line 883, in rosbagmain
    cmds[cmd](argv[2:])
  File "/tmp/robag/src/ros_comm/tools/rosbag/src/rosbag/rosbag_main.py", line 347, in filter_cmd
    for topic, raw_msg, t in inbag.read_messages(raw=True):
  File "/tmp/robag/src/ros_comm/tools/rosbag/src/rosbag/bag.py", line 2333, in read_messages
    yield self.seek_and_read_message_data_record((entry.chunk_pos, entry.offset), raw)
  File "/tmp/robag/src/ros_comm/tools/rosbag/src/rosbag/bag.py", line 2472, in seek_and_read_message_data_record
    msg_type = _get_message_type(connection_info)
  File "/tmp/robag/src/ros_comm/tools/rosbag/src/rosbag/bag.py", line 1566, in _get_message_type
    message_type = genpy.dynamic.generate_dynamic(info.datatype, info.msg_def)[info.datatype]
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genpy/dynamic.py", line 160, in generate_dynamic
    for l in msg_generator(msg_context, spec, search_path):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genpy/generator.py", line 736, in msg_generator
    genmsg.msg_loader.load_depends(msg_context, spec, search_path)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 344, in load_depends
    return load_msg_depends(msg_context, spec, msg_search_path)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 313, in load_msg_depends
    depspec = load_msg_by_type(msg_context, resolved_type, search_path)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 119, in load_msg_by_type
    file_path = get_msg_file(package_name, base_type, search_path)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 78, in get_msg_file
    % (base_type, package, search_path))
genmsg.msg_loader.MsgNotFound: Cannot locate message [TransformStamped]: unknown package [geometry_msgs] on search path [{}]`

dirk-thomas, 2017-04-25:
Please provide more information: what exact versions are you using? What is your environment which doesn't seem to include `geometry_msgs`? Please provide the exact steps to reproduce the problem (including e.g. the bag files used of necessary).

dirk-thomas, 2018-02-20:
@themightyoarfish A "me too" comment isn't going to help for this issue. Please provide the additional information asked for.

themightyoarfish, 2018-02-21:
It later turned out that it appears the bagfile was the problem since with another one containing the same message types there was no problem. Still it is strange because the bag could be played back with no problem.

closed 2025-05-31 by sloretz