ros/ros_comm #1039: Completely disable logging outputs on screen

inspire1989, 2017-04-27:
Hi, 
I want to disable the ROS logging in the console completely, but still have the logging to the /rosout topic. I tried these two options:

1. Set output="log" in the launch file. This disables the output of Debug, Info and Warning messages but still, Error and Fatal messages are shown.
2. Set the logger level to Fatal but this completely disables the /rosout topic.

Are there any other possibilities I could try out? For me, it sounds like a design-issue, because, if I explicitly specify the "log" output, I do not want to have outputs in the console...

Otherwise, I would be grateful to see this fixed in the next ROS version! :)

Thanks!

inspire1989, 2017-04-27:
By the way, I have another logging framework running that should still be able to log everything to the console.

dirk-thomas, 2017-04-28:
In ROS the log threshold determines that all log messages with a lower level are being ignored. The backend actually performing the logging can be swapped (see existing implementations: https://github.com/ros/ros_comm/tree/lunar-devel/tools/rosconsole/src/rosconsole/impl).

You use case of assigning separate thresholds for different sinks isn't supported. You might be able to configure / extend the log4cxx backend to match your needs.

But this is likely not a feature the maintainers will work on in the future. Please consider to contribute a pull request if this functionality is useful for you.