ros/ros_comm #1047: Rosbag API is not compatible with Python 3

nstiurca, 2017-05-08:
Due to changes in handling of files with regards to `str` vs `bytes` between Python 2 and 3, the `rosbag` API doesn't work in Python 3. I have a very simple, elegant fix which I will set up a PR for shortly, and I hope it can be included in Kinetic.

For reference, my system is Mint 18.1 (ie Ubuntu 16.04) running Kinetic, and we definitely need to use Python 3 for most of what we do.

closed 2017-08-23 by dirk-thomas

ehannigan, 2017-10-16:
I apologize if this is the wrong place to ask this question, but was this fix added to kinetic? Both merges show this only being added to ros:Lunar. It would be really useful to have in kinetic. 

If so, how do I go about getting the update? 

Thank you,
E

dirk-thomas, 2017-10-16:
@ehannigan All fixes are being committed to the branch of the latest ROS distro first. Once they have been released into the latest ROS distro (which isn't the case for #1150 yet) and that hasn't resulted in any regressions the patch is being considered for backporting into older ROS distros.

You can either wait for this to happen and become available from Debian packages in Kinetic or you can build the specific package from source and apply the same patch locally.

ehannigan, 2017-10-16:
@dirk-thomas 
Thanks for the quick response! I will look into installing the patch locally.