ros/ros_comm #1051: Add substitution args parsing to roslaunch yaml loaded from file

DSsoto, 2017-05-09:
@jbohren's #218 added resolving args in rosparam if the yaml text is inlined inside of the launch file. 

However I have repeatedly found myself missing this feature when loading parameters from a external yaml file using `rosparam`'s `file` attribute. Here is a minimal reproducible example of the issue.

config.yaml:
```
# This arg will not get substituted corrctly
subbed_pkg_from_file : $(find roscpp)
```

param_loader.launch:
```
<launch>
  <rosparam subst_value="true"> subbed_pkg : $(find roscpp) </rosparam>
  <rosparam subst_value="true" file="/home/santiago/config.yaml" />
</launch>
```

Output of `roslaunch param_loader.launch --dump-params`:
> {/subbed_pkg: /opt/ros/kinetic/share/roscpp, /subbed_pkg_from_file: $(find roscpp)}

It would be very useful to actually have `subbed_pkg_from_file` resolve to the package path when `subst_value` is set to true.

dirk-thomas, 2017-05-09:
The maintainers will likely not have time to implement this in the foreseeable future. Please consider to contribute a pull request to implement this feature.

dirk-thomas, 2018-08-20:
Implemented in #1354.

closed 2018-08-20 by dirk-thomas