ros/ros_comm #1053: Possible Floating point precision limitation in rosbag.readmessages timestamps

PeterQFR, 2017-05-10:
When using rosbag in python as in the following there is a precision issue which limits the digitisation of the timestamps that function produces such as in the following code:

```
with rosbag.Bag(filename) as bag:
        statevector =[];
        timestamps =[];
        i=0
        
        for topic, msg, timestamp in bag.read_messages(topics=[topic]):
            ...timestamp cannot show a precision below 50ms
```
Its effect is that timestamps are grouped until the next discretization step. such that a message that is published at a frequency of 100Hz will have groups of 5 messages that have the same timestamp. 

This is inherent in the representation of the rosbag.time structure as printing out the time.nsecs value shows that these values are all the same. 

This is a problem with the way rosbag assigns the timestamp objects, **_possibly due to a conversion somewhere using a floating point representation_**, and/or the fact **_that the rosbag records time from the epoch start as opposed to the bag start_**, where floating point representations would experience precision issues. 

Note the msg.header.stamp values are correct.

dirk-thomas, 2017-07-14:
Depending on which format version your bag file uses the reading happens in different locations:

* https://github.com/ros/ros_comm/blob/bdc8c3d66334e092590730bc88e6b2adcff59b08/tools/rosbag/src/rosbag/bag.py#L1832-L1835
* https://github.com/ros/ros_comm/blob/bdc8c3d66334e092590730bc88e6b2adcff59b08/tools/rosbag/src/rosbag/bag.py#L2051-L2054
* https://github.com/ros/ros_comm/blob/bdc8c3d66334e092590730bc88e6b2adcff59b08/tools/rosbag/src/rosbag/bag.py#L1586

But these locations all have in common that they read two integers (one for `secs` and one for `nsecs`). Also the later used data structure `genpy.Time` uses two separate integer fields. Therefore I don't see where this rounding / truncating should come from.

Please provide more information about the problem as well as a [SSCCE](http://sscce.org/).

PeterQFR, 2018-01-16:
I think this is might be a rosbag issue, in the way that messages are buffered and written during a record. Sometimes groups of messages will be written in small bursts.  In using the message timestamps, (not those provied from timestamp in the rosbag.read_messages()), this issue goes away.

closed 2018-01-16 by PeterQFR