ros/ros_comm #1059: message_filters: time synchronizers should print warning about dropped messages

mintar, 2017-05-18:
It would be nice if the approximate and exact time synchronizers would print a warning when they are dropping all the messages (for example, because one of the synchronized topics is unconnected). Something similar to [what tf::MessageFilter does](https://github.com/ros/geometry/blob/4afac63990b2b8ce04860f4313068b61fd36815a/tf/include/tf/message_filter.h#L496-L497); that warning has often saved me hours of debugging.

dirk-thomas, 2017-05-18:
That sounds indeed useful. Please consider to provide a pull request for this.

ashwinvkNV, 2023-04-23:
Has this been implemented or is there any work in progress?

mintar, 2023-05-10:
Not as far as I know. Still looking for volunteers to provide a pull request!

closed 2025-05-31 by sloretz