ros/ros_comm #1063: Better error output when logging cannot create folder

nordmoen, 2017-05-23:
Summary:
In the function [`configure_logging`](https://github.com/ros/ros_comm/blob/247459207e20c1da109fc306e58b84d15c4107bd/tools/rosgraph/src/rosgraph/roslogging.py#L86) it would be nice if in addition to printing that the folder cannot be created print the exception being thrown.

Longer explanation:
I am having a problem where I need to run ROS on a limited environment. To allow for logging the system lets me create a folder and expose that to ROS. The problem I have been having is that when the system creates the folder it is created with different group than ROS is running under. This resulted in the situation where ROS was able to create the logging directory, but not change the folders permissions (parent folder had group `root`). I have used quite some time debugging why this was happening and it took some time before I found the function responsible since the folder was created successfully, but ROS still stopped with "cannot create directory".

dirk-thomas, 2017-05-24:
That sounds like a good idea. Please consider to provide a pull request for this improvement (against the latest development branch, currently `lunar-devel`).

dirk-thomas, 2017-07-07:
Addressed by #1068.

closed 2017-07-07 by dirk-thomas