ros/ros_comm #1064: Should roslaunch param tag attribute command strip whitespace?

sloretz, 2017-05-23:
roscore.xml creates two parameters `rosversion` and `rosdistro`. These are created using the roslaunch `param` tag with the attribute `command`. `rosparam dump` shows these attributes have a newline in them.

```
$ rosparam dump
rosdistro: 'kinetic

  '
roslaunch:
  uris: [...]
rosversion: '1.12.7

  '
```

Should the whitespace be stripped from these parameters? How does roslaunch execute the commands? It doesn't seem to run it in a shell because pipes and back-ticks have no effect.

Observed in kinetic and lunar. cc @mikaelarguedas

dirk-thomas, 2017-05-24:
`roslaunch` will simply return the exact output of the command. And the command does include the trailing newline (see ros-infrastructure/rospkg#26) for all ROS distros.

Since the tool `rosversion` is also used from the command line it is "good" that it includes the trailing newline.

`roscore` could invoke the command with a new option to leave the trailing newline out or somehow apply the trimming to the output of the command.

mikepurvis, 2017-05-29:
We absolutely do this trick in a bunch of our launch files, and having to strip off the whitespace is an annoying added hassle. I'd support an explicit attribute that strips whitespace (similar to how CMake does), eg:

    <param name="rosversion" command="rosversion [pkgname]" strip="true" />

Alternatively, this could be handled within `rosversion`, either by:

- A flag which suppresses the final newline, or
- It simply recognizing when it is being launched non-interactively, and suppressing the newline in that case.

dirk-thomas, 2017-07-07:
Please consider to contribute a pull request for either of the two proposed solutions.

closed 2025-05-31 by sloretz