ros/ros_comm #1066: rosparam: pipe character in pretty print makes dump to file unusable for xml checking

guihomework, 2017-05-23:
It happened several times I had to check a generated urdf with ```check_urdf``` or ```urdf_to_graphviz```

To get the generated urdf I found two options :
 * Find the upload.launch file where the robot_description is generated, convert that xml line into a command line calling xacro and replacing the $(find) in rospack find , piping the output to a file and passing it to check_urdf
 * A faster way was to launch this upload.launch and get the generated urdf from the param server and then load it in check_urdf with 
``` 
     rosparam get /robot_description > /tmp/myrobot.urdf 
     check_urdf  /tmp/myrobot.urdf 
```
The second option does not work as is, due to double quotes as well as end-of-line escaped everywhere in the output. 
```
"<?xml version=\"1.0\" ?>\n   ... "
check_urdf /tmp/myrobot.urdf 
Error:   Error document empty
```
Fine. I used pretty print option (-p)  before storing to the file, but it does not work either due to a **pipe** character '|' at the start of the output. 

```
|
  <?xml version="1.0" ?>
check_urdf /tmp/myrobot.urdf 
Error:   Error document empty
```
Removing this pipe in the first line solves the issue and check_urdf is happy

So my question is simple, why is there a pipe here https://github.com/ros/ros_comm/blob/lunar-devel/tools/rosparam/src/rosparam/__init__.py#L270  in the pretty print ? 

I suppose it is for pretty visual alignment but I don't remember I have seen an example of rosparam output using it.

At least if the content is a single string, why should there be a pipe at the start ?

BTW I also tested to use rosparam dump  which produces the same escaped string as with no pretty print.

dirk-thomas, 2017-05-24:
That is hard to say. That code part was written before the code got into git (7+ years ago). I would guess it is being done to follow the yaml style of multi-line strings (see http://www.yaml.org/spec/1.2/spec.html#id2761083). The reason could be to allow passing the output directly to other scripts reading yaml input.

guihomework, 2017-05-24:
Hi, 
thanks for having taken the time to look at this. I also understand this old code, and finding reason of certain choices is tricky now.

To see if spec plays a role I wrote a foo.yaml containing this example 
```
example: 
  name: Mark McGwire
  accomplishment: >
    Mark set a major league
    home run record in 1998.
  stats: |
    65 Home Runs
    0.278 Batting Average
```

loading it on the param server lists this way
```
/example/accomplishment
/example/name
/example/stats
```
and outputs this way with rosparam get /example
```
{accomplishment: 'Mark set a major league home run record in 1998.

    ', name: Mark McGwire, stats: '65 Home Runs

    0.278 Batting Average

    '}
```
and this way with pretty print
```
accomplishment: |
  Mark set a major league home run record in 1998.
  
name: Mark McGwire
stats: |
  65 Home Runs
  0.278 Batting Average
```
which respects the spec indeed but from a very quick test, I replaced the pipe character at the line 270 with a 'Z' and it did not affect the output of this example and any other example of the spec including the pipe character. The reason is that this part is processed here https://github.com/ros/ros_comm/blob/lunar-devel/tools/rosparam/src/rosparam/__init__.py#L234

However my robot_description had a 'Z' at start now.
```
 rosparam get /robot_description -p
Z
  <?xml version="1.0" ?>
```
So there should be another reason than style.  I don't understand why a character like this could help passing to application on stdin for instance. 

So if anybody has an example of yaml that needs this character at the beginning, we could find a work around in the specific case of a single string like xml.

guihomework, 2017-05-24:
Hum, I think I found the reason of this character thanks to this explanation https://stackoverflow.com/questions/15540635/what-is-the-use-of-pipe-symbol-in-yaml

if I do this in ```ipython``` I get the correct input back with indentation
```python
In [1]: import yaml
In [2]: a = yaml.load("""
   ....: |
   ....:   <?xml version="1.0" ?>
   ....:     <robot>
   ....:     </robot>
   ....: """)

In [3]: print a
<?xml version="1.0" ?>
  <robot>
  </robot>
```
but without the initial pipe, we loose the identation and end of lines.
```python
In [4]: b = yaml.load("""
   ....: <?xml version="1.0" ?>
   ....:   <robot>
   ....:   </robot>
   ....: """)

In [5]: print b
<?xml version="1.0" ?> <robot> </robot>
```

I suppose there are a lot of load/dump of yaml everywhere in ```roslaunch``` etc, however I am not sure the pretty print option is used there, and then this pretty print could be fixed if it does not affect the internals of ROS regarding param/yaml load/dump.

```check_urdf``` is ros agnostic so cannot have a mode to directly get the param from param server. It is expecting a well formed xml, and one should not make a specific case for malformed yaml dump of xml content. 

I guess the simplest and only choice I have is to write my own bash script with an alias that dump the param to a file, removes the pipe and calls ```check_urdf``` afterwards.

closed 2025-05-31 by sloretz