ros/ros_comm #1070: roslaunch: "multiple files named ... in package" / inconsistent finding of resources

fschopp, 2017-06-01:
The current [roslaunch package summary](http://wiki.ros.org/roslaunch) seems to indicate that the standard way of using roslaunch on the command line is `roslaunch package_name file.launch`. This creates problems when a launch file exists both in the source tree and in the devel tree.

```
$ roslaunch bar foo.launch
multiple files named [foo.launch] in package [bar]:
- /workspace/devel_isolated/bar/share/bar/foo.launch
- /workspace/src/bar/src/main/launch/foo.launch
Please specify full path instead
The traceback for the exception was written to the log file
```

To work around this, one could refrain from copying launch files to the devel space (and instead only have them in the source and in the install space). But if I delete /workspace/devel_isolated/bar/share/bar/foo.launch, then `$(find bar)/foo.launch` in a dependent launch file no longer resolves correctly.

```
$ roslaunch baz toplevel.launch
... logging to /some/file/path.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt

while processing /workspace/src/baz/toplevel.launch:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/workspace/src/bar/foo.launch'
The traceback for the exception was written to the log file
```

So the workaround (i.e., not having launch files in the devel space) only helps if the location of the launch file relative to the package root is the same in the source and in the install space.

The underlying issue seems to be that resource finding is inconsistent in roslaunch. The two-argument form of roslaunch uses [`roslib.packages.find_resource`](https://github.com/ros/ros_comm/blob/1.13.0/tools/roslaunch/src/roslaunch/rlutil.py#L92). However, `$(find pkg)` expressions are resolved using [`catkin.find_in_workspaces`](https://github.com/ros/ros_comm/blob/1.13.0/tools/roslaunch/src/roslaunch/substitution_args.py#L217-L219) (called from [`substitution_args._find`](https://github.com/ros/ros_comm/blob/1.13.0/tools/roslaunch/src/roslaunch/substitution_args.py#L156-L158)).

dirk-thomas, 2017-06-02:
> This creates problems when a launch file exists both in the source tree and in the devel tree.

First of all you shouldn't copy any files from the source space to the devel space. The idea of the devel space is to avoid that step. Otherwise you could simply do an install step.

> `$(find bar)/foo.launch` in a dependent launch file no longer resolves correctly.

`$(find bar)` can't handle this since it can only return a single path for the package `bar`. But in order to find a specific file in either of the two possible locations the filename must be considered too. (See #228 for a lengthy discussion and past proposals which no consensus could be reached on.)

A workaround has been implemented in #233 which should make the following (`$(find bar)/launch/foo.launch`) work if you make sure to have the file:
* in the source space under `<source space of pkg>/launch/foo.launch` and
* installed to `<install prefix>/share/<pkg name>/launch/foo.launch`

dirk-thomas, 2017-07-21:
I will close this ticket since I assume my comment resolved your problem. Please feel free to comment and the ticket can be reopened if necessary.

closed 2017-07-21 by dirk-thomas