ros/ros_comm #1073: publish_single in rospy SubscribeListener.peer_subscribe is inconsistent

BenjaminPelletier, 2017-06-09:
Run roscore.  In another terminal, run the script below.  In a third terminal, run 'rostopic echo /test_topic'.  Observe that it (probably) echoes both messages (p1 and p2).  Ctrl-c then run again.  Repeat as many times as necessary to observe an occurrence where only one of the two messages is echoed.  For me on ROS Kinetic with Ubuntu 16.04 on a pretty powerful machine, it generally takes 1-5 trials to get an echo with only one message.  When a message is missing, it always seems to be p1 (p2 always gets echoed).

The printing statements indicate that the node always thinks the pseudo-latched message is published to the new subscriber (publish_single is invoked for both messages).  This seems to mean that publish_single (called peer_publish in SubscribeListener.peer_subscribe) sometimes fails silently.  That seems pretty unacceptable.

For context, I'm trying to work around the 4-year-old bug 146.

    import rospy
    from std_msgs.msg import String
    
    lpid = 0
    class LatchPublisher(rospy.Publisher, rospy.SubscribeListener):
        def __init__(self, name, data_class, tcp_nodelay=False, headers=None, queue_size=None):
            global lpid
            self.id = lpid
            lpid += 1
            print("Created LatchPublisher %d %s" % (self.id, name))
            super(LatchPublisher, self).__init__(name, data_class=data_class, tcp_nodelay=tcp_nodelay, headers=headers, queue_size=queue_size, subscriber_listener=self, latch=False)
            self.message = None
    
        def publish(self, msg):
            print("Publishing message on LatchPublisher %d" % self.id)
            self.message = msg
            super(LatchPublisher, self).publish(msg)
    
        def peer_subscribe(self, resolved_name, publish, publish_single):
            print("A subscriber attached to LatchPublisher %d %s" % (self.id, resolved_name))
            if self.message is not None:
                print("  Publishing 'latched' message %d" % self.id)
                publish_single(self.message)
                print("  Published 'latched' message %d" % self.id)
    
    rospy.init_node('test_latch_publisher')
    
    print("Creating LatchPublishers")
    p1 = LatchPublisher("/test_topic", String, queue_size=1)
    p2 = LatchPublisher("/test_topic", String, queue_size=1)
    
    print("Latching messages")
    p1.publish(String('p1'))
    p2.publish(String('p2'))
    
    print("Spinning")
    rospy.spin()

evenator, 2017-06-09:
In rospy, do multiple asynchronous publishers to the same topic share a queue? That's counter-intuitive, but it would explain this behavior for `queue_size=1`.

If that's the case, which size takes precedence in the case of two publishers with different queue sizes?

BenjaminPelletier, 2017-06-09:
That does indeed seem to be the issue -- when I change queue_size to 10, I can't reproduce the problem.  In that case, the bug is that buffer overruns in Publisher do not produce error messages.

Regarding the queue_size for two publishers, empirically the queue_size of the singleton appears to be simply overwritten.  Setting p1 size to 1 and p2 size to 10 results in no problems.  Setting p1 size to 10 and p2 size to 1 does result in problems.

BenjaminPelletier, 2017-06-09:
To clarify the new bug title, the reason this was a problem is that the rospy.Publisher was overwriting an unprocessed queue element silently so the overrun was not detected.  When the queue overflows, it should produce some kind of error at some level in the system.

dirk-thomas, 2017-07-07:
> Do multiple asynchronous publishers to the same topic share a queue?

Yes, all `Publisher` instances on the same topic share the same `_PublisherImpl` singleton.

> If that's the case, which size takes precedence in the case of two publishers with different queue sizes?

The last created publisher will override the setting of the singleton.

The observed behavior is actually exactly how the functionality is being implemented. While I see your use case the behavior you are seeing is not actually a bug but the implementation choice made in the early days of ROS.

> When the queue overflows, it should produce some kind of error at some level in the system.

In many use cases the queue is being used to limit the amount of messages it can store. Intentionally there is no error message being printed in these cases. Adding such a message now would be unexpected in many existing applications therefore I don't think we can add this.

evenator, 2017-07-07:
> The last created publisher will override the setting of the singleton.

I would strongly recommend emitting a warning when someone overrides a previous queue length, as the behavior is opaque right now.

BenjaminPelletier, 2017-07-07:
> Yes, all Publisher instances on the same topic share the same _PublisherImpl singleton.

This is the root of much evil (see, for instance, issue 146).

> not actually a bug but the implementation choice

Bugs are merely implementations that don't produce the desired behavior.  It seems unlikely that silently mutating the behavior of a separate, unreferenced object is really a desirable behavior.  Instead, it seems like an undesired side effect of incompletely optimizing Publisher performance.

> Adding such a message now would be unexpected in many existing applications

Allowing developers to rely on undocumented behaviors (the silent discarding of buffer overruns) that turn out to cause undesirable consequences doesn't seem like a great way to produce high-quality software, but if we take that as a given that we can't change, at least adding an optional callback to be invoked upon buffer overrun would allow those of us who don't like silently ignoring problems to catch them effectively.

dirk-thomas, 2017-07-07:
> I would strongly recommend emitting a warning when someone overrides a previous queue length, as the behavior is opaque right now.

@evenator Any code using more than one publisher for the same topic currently uses this. In which cases would you like to see a warning? If the value is different? If the value is smaller? What if the value is actually increased - will that be enough because caller planned for this or does the user still need a warning?

@BenjaminPelletier I was merely describing how the current code works which I haven't implemented myself as well as laid out the problems in changing that behavior now. The optional callback is certainly an option. Please consider to provide a pull request to implement that enhancement.

evenator, 2017-07-08:
> @evenator Any code using more than one publisher for the same topic currently uses this. In which cases would you like to see a warning? If the value is different? If the value is smaller? What if [...]

I think that if the value is different, a warning is warranted. This is what I'd call "spooky action at a distance." The developer-facing API implies that `Publisher`s are independent objects, so there's no reason to assume that a second `Publisher` even *could* affect the first.

I can't imagine a situation where someone would intentionally override the queue length of an existing `Publisher` when they create a second `Publisher` on a topic. There may be a use-case, so disabling the functionality would be disruptive, but I think anyone doing it is *probably* doing so by accident and may suffer unintended consequences. A warning is a good way to tell them that they probably did something they didn't mean to. If they know what they're doing, they can ignore it.

dirk-thomas, 2017-07-08:
> I can't imagine a situation where someone would intentionally override the queue length of an existing Publisher when they create a second Publisher on a topic.

One use case I can imagine is when a user is aware of the current behavior and has code which registers multiple publishers. Every time a publisher is registered the desired `queue_size` is added to a global variable. That way with each additional publisher the queue size is appropriately increased. E.g. publisher A needs a queue size of 3 and publisher B needs 5. Then when B is registered a queue_size of 8 is passed to achieve the desired goal. How will that user be able to use the API without a warning being printed? Just telling users they have to ignore the warning message is in my opinion not viable. I think there needs to be a way to use the API in a way that it doesn't print warnings.

evenator, 2017-07-08:
The [ROS wiki page](http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers) and the [API documentation](http://docs.ros.org/api/rospy/html/rospy.topics.Publisher-class.html) should refer to this behavior.

pjreed, 2017-07-08:
> One use case I can imagine is when a user is aware of the current behavior

I'm having a hard time imagining how a user could even be aware of this current behavior.  It's not documented and completely non-intuitive that parameters to an object's constructor would change parameters on a global singleton.  I'd much rather have that parameter completely removed from the constructor and have it set via a static method, but that's probably not a good idea since it would break the current API.  I would be shocked if anybody is actually intentionally using this behavior.

In the meantime I think it'd be reasonable to have it log a warning when the behavior is triggered and then add another parameter or variable that can be set to suppress it in case that's what somebody really wants.

dirk-thomas, 2017-07-10:
> The ROS wiki page and the API documentation should refer to this behavior.

Please feel free to update the wiki directly and/or provide a PR to improve the docblock. The subscriber already contains some explanation in the API which might be a good starting point (see [1](https://github.com/ros/ros_comm/blob/c60e52ec75e786f6e9515f60e20cdb6ee37c9f6e/clients/rospy/src/rospy/topics.py#L519-L524) and [2](https://github.com/ros/ros_comm/blob/c60e52ec75e786f6e9515f60e20cdb6ee37c9f6e/clients/rospy/src/rospy/topics.py#L541-L547)). While I would be happy to do so myself it might be quite a while until I will have time for it.

BenjaminPelletier, 2017-07-10:
Unless you really think it's a good design intent to silently mutate unreferenced objects, I suggest that this is a bad behavior.  Bad behavior is a bug, and bugs should be fixed rather than documented.

dirk-thomas, 2017-07-10:
> Unless you really think it's a good design intent to silently mutate unreferenced objects, I suggest that this is a bad behavior. Bad behavior is a bug, and bugs should be fixed rather than documented.

If I would write this code from scratch I would certainly consider a different approach. But that option is not on the table. This logic has been implemented a long time ago and a lot of code has been implemented on top of it in the mean time. So backwards compatibility is a big concern here.

I don't think adding a warning or error message for every code using this which has worked fine until now for the developers is a reasonable thing. Also requiring all existing code to be updated to pass a new flag to silence the message is something I don't think is reasonable to expect.

Under these constraints I think making the fact how this option works more visible is a good first step. As mentioned before I am always open for concrete proposals how to improve this further.

evenator, 2017-12-01:
As far as I can tell, this remains completely undocumented on the wiki and the API docs, almost five months later.

wjwwood, 2017-12-01:
@evenator you can edit the wiki and make pr's to the doc block too. It would help us a lot if you or anyone else from this thread could add notes about it in the docs.

----

While I agree with you and the others in this thread that the behavior is undesirable, @dirk-thomas's point about maintaining stable behavior is valid. A user doesn't need to be aware of the exact behavior to be accidentally relying on the current behavior. The change in behavior would bring with it substantial (in my opinion) code changes which might bring new bugs with it, potentially further disrupting existing software. So, as the maintainers we have to be extra careful about changes like this.

We could change this in a subsequent ROS release though. I think either a change in behavior or a warning (so long as there is a work around to suppress the warning in unavoidable cases) would be appropriate in future ROS releases.

We already have a tough time keeping up with pull requests and high priority issues related to security and licensing for the core ROS packages, so when we ask you guys to consider opening a pull request for these changes or add something to the documentation, it's not that we don't care about it, but rather that we simply don't have time for everything and could use some help.

dirk-thomas, 2017-12-19:
Please feel free to create a PR for this change now / soon. Once we have a melodic branch it will be considered to be merged for the next ROS distro.

dirk-thomas, 2020-08-03:
With Noetic being released and likely being the last ROS 1 distro and this ticket proposing to change behavior which shouldn't target and already released distro I will go ahead and close it.

closed 2020-08-03 by dirk-thomas