ros/ros_comm #1078: Topic subscription scalability problem

guillaumeautran, 2017-06-12:
Looking into understanding the performance of rosbag recorder, I stumble over something which seems to be a performance bottleneck. The actual usecase is to record about >300 topics with varying publication rate. The test setup used for performance measurement and profiling is a single rosbag recorder recording 309 active topics over a period of 2 minutes with no compression. The run is profiled using VTune and profiler data analyzed.

The performance of the existing topic monitoring is shown in the following graph:  
[image]
Based on those findings, the top CPU usage of the rosbag recorder goes to the ros::PollSet class. This class is polling all file-descriptors which is known to be an inefficient strategy when monitoring over 100 sockets. The poll system call handles descriptors in a linear way. (source: https://daniel.haxx.se/docs/poll-vs-select.html)

In addition to my investigation, I've also implemented a event based implementation (using libev) as a comparison and also profiled the application which yield much better performance as seen in the graph: 
[image]

So, I'd like to ask if anyone has already looked into asynchronous topic socket monitoring and if not, would a modification like the one proposed be of interest?

mikepurvis, 2017-06-13:
(Guillaume is a colleague at Clearpath)

Looks like the current implementation is here: https://github.com/ros/ros_comm/blob/lunar-devel/clients/roscpp/src/libros/poll_set.cpp

Which is ultimately backed by the `poll_sockets` function here: https://github.com/ros/ros_comm/blob/247459207e20c1da109fc306e58b84d15c4107bd/clients/roscpp/src/libros/io.cpp#L104

There's a pretty extensive Win32-specific implementation already in there, so I'd be nervous about proposals to switch wholesale to a cross-platform solution such as `libev` or `boost::asio`. However, adding a compile-time switch that uses the [epoll functions](https://linux.die.net/man/4/epoll) instead of poll seems like it could be a minimally-invasive option that would improve the situation on epoll-supporting platforms (checking the availability of epoll from CMake is [easy to do](https://github.com/skroll/cmake-modules/blob/master/modules/CheckEpoll.cmake)).

@dirk-thomas Preferences on how to move this forward?

dirk-thomas, 2017-07-07:
I agree with @mikepurvis statement. Switching to `libev` seems to be a huge change with a lot of potential for regressions. Therefore chances for getting that merged into existing ROS distros is pretty low. Changing to epoll on the other hand should be much easier and less invasive. For other parts of ROS that has already been done (#831) or attempted (#833, this change needed to be reverted for now due to unforseen regressions). It would be great if you could provide a PR for this improvement - it will likely require quite some testing but the scalability impact would be great!

guillaumeautran, 2017-07-07:
Thanks @dirk-thomas for the feedback. I was thinking to use `boost::asio` which under the cover used `epoll` on platform that support it. Using the `boost::asio` library would also make it windows compatible avoiding special code for that platform. 

Yes, I'll get working on a PR

PR #1217 (2017-12-21): Topic subscription scalability fix
Switching to using epoll system calls to improve performance of the topic polling code by a factor of 2.

See previous discussions and rationale in #1162 and https://github.com/clearpathrobotics/ros_comm/pull/12. Closes #1078.
closed 2017-12-21 by mikepurvis