ros/ros_comm #1082: Proposal: shm transport native support in ROS

quning78, 2017-06-23:
Hi, All,

We are trying to using ROS in some areas with new requirements:

* potential huge amount of input data,
* high reliability.

That's why we come back to this topic again. I know there are previous discussion about this topic, e,g. 

http://ros-users.122217.n3.nabble.com/Shared-memory-transport-using-C-in-ROS-td414682.html

However, I would like to reiterate some reasons based the current circumstance.

* with the requirements listed above, we have nodelet for first. But with thread model, the reliability of nodelet is much worse than node.
* why ros instead of ros2? we know that ros2 support shm transport pretty well with DDS integration. However, ros is so popular in the world and ros2 has yet to come.

So, we would like to contribute the design and implement for this feature if you guys are interested in this.

wjwwood, 2017-06-23:
> But with thread model, the reliability of nodelet is much worse than node.

Why do you say this? Because of process isolation or something else?

quning78, 2017-06-23:
Yes, because of the isolation is better for process compared to thread.

wjwwood, 2017-06-23:
The idea is reasonable, we have just never had enough reason to do it because in our experience only the latency is affected by using shared memory over TCP. There was some work done by some students comparing shared memory with loopback TCP to support that idea. I just pinged them to see where that work landed, but I think the consensus was that it is better than TCP, but only marginally and only really for latency but not really for throughput.

As I said, that's been the prevailing idea, but other than this paper I'm trying to track down for you, there's not a lot of hard evidence for this.

But that only means we never saw a strong need for it and so we didn't work on it. I think it would be fine to integrate into `roscpp` if someone like yourself is interested and motivated in doing so.

I'd wait for feedback from @dirk-thomas and the other @ros/ros_comm-maintainers before starting work though.

cbames, 2017-06-23:
Here's the report that @wjwwood  was referring to.  What we found is that a large number of small messages are better served by SHM transport, but large messages were actually transported faster over TCP. 

As far as reliability goes, we found that very large messages (over 50mb each), weren't well served by TCP. It's unclear why this is the case, it might not be an inherent problem, but rather a limitation imposed by ROS on TCP. There was a small hit in reliability for very small messages, which would also require more investigation. We didn't investigate any further because reliability was above 95%.

[Project report.pdf](https://github.com/ros/ros_comm/files/1099211/Project.report.pdf)

quning78, 2017-06-23:
Thanks for all your inputs.

The performance gain comes different sides
* as you mentioned. Shm over TCP maybe not that much for small scale msg, but the gain increases dramatically with msg size increased. . 
* also the number of receiver make the copies of the message increased as well. E.g.

quning78, 2017-06-23:
Thanks for the inputs. I will take a look.

quning78, 2017-06-23:
Btw, not sure why this is marked as closed.

dirk-thomas, 2017-07-06:
Adding an additional optional transport using shm should be fine to be added in ROS 1. As long as the default behavior is not changed I only see minimal chances for regressions. So if you are willing to contribute a patch for this feature we will certainly be happy to review / merge it.

quning78, 2017-07-06:
That's great to know! We will prepare for the design review and the patchset pretty soon.

Jrdevil-Wang, 2017-07-15:
Really glad that we find this issue. We've been working on this topic for a while, and it is a great pleasure to know someone who is also interested. @quning78 
I have not dig into the project report that @cbames mentioned yet, but our results are the opposite.
Our shm_transport is suitable for multi-subscriber and large messages.
Here is the current version, https://github.com/Jrdevil-Wang/shm_transport.
There are still some FIXME, and very welcome to offer improving advices.

Jrdevil-Wang, 2017-07-24:
We have added some comparison results.
https://github.com/Jrdevil-Wang/shm_transport/blob/master/Latency%20vs%20Message%20Size.pdf
https://github.com/Jrdevil-Wang/shm_transport/blob/master/Latency%20vs%20Number%20of%20Listeners.pdf
The latency can be greatly improved, especially when the number of listeners increases.

I think the main difference between our implement and the report which @wjwwood mentioned is the notification method. In the report, each listener creates a thread to check if there are new messages. While in our implement, we use a std_msgs::UInt64 message to notify listeners when there are new messages. The comparison results show that our implement is more scalable.

Is this suitable for ROS 1? How can we contribute a patch? @dirk-thomas

dirk-thomas, 2017-07-24:
The results look very good indeed.

> Is this suitable for ROS 1? How can we contribute a patch?

As I mentioned before:

> Adding an additional optional transport using shm should be fine to be added in ROS 1. As long as the default behavior is not changed I only see minimal chances for regressions.

So if you would like to create a PR please create it against the latest branch (currently `lunar-devel`). Please avoid any unrelated changes to keep the patch minimal. The patch should not change any public API and shouldn't change the default behavior (a user would like choose the new transport explicitly using an optional argument). The patch should also add tests which cover the new transport. Please try to keep unrelated changes in separate commits since that will help reviewing the individual parts of the patch.

quning78, 2017-07-24:
We would like to help on this as well since we have some code already on
this feature.

Best wishes,
-- 
Ning Qu

Jrdevil-Wang, 2017-07-28:
I just create a pull request https://github.com/ros/ros_comm/pull/1112.
It does not contain any tests yet, since I am still trying to figure out how gtest works.
I am still new to ROS. Please let me know if there are any other issues need to be taken care of. @dirk-thomas

Jrdevil-Wang, 2017-11-01:
Hi, I just drawback PR #1112. It seems that it has been ignored.
I am not complaining. I simply feels that this version is not very suitable to be a part of ros_comm, since it changes the API a little bit. I will try to get there though.
Meanwhile, I have submitted a package, and it have been indexed. http://wiki.ros.org/shm_transport
Very welcome to give any advice!

dirk-thomas, 2017-11-01:
@Jrdevil-Wang just because the PR hasn't received any comments doesn't mean it has been "ignored". Looking at the patch now it seems it adds a few files but they are not being used anywhere. The PR also doesn't contain any more information how this is supposed to be used. That might be one reason why it hasn't received any attention.

Jrdevil-Wang, 2017-11-02:
Thanks for replying @dirk-thomas . I see your point, and I will try to do better next time (hopefully soon :D).

closed 2025-05-31 by sloretz