ros/ros_comm #1089: [roslz4] inconsistent usage of paths

Alessandro-Barbieri, 2017-07-05:
I can't get the .so installed alongside the .py
on my gentoo I have roslz4 installed in

```
/opt
/opt/ros
/opt/ros/include
/opt/ros/include/roslz4
/opt/ros/include/roslz4/lz4s.h
/opt/ros/lib64
/opt/ros/lib64/libroslz4.so
/opt/ros/lib64/pkgconfig
/opt/ros/lib64/pkgconfig/roslz4.pc
/opt/ros/share
/opt/ros/share/ros_packages
/opt/ros/share/ros_packages/roslz4
/opt/ros/share/ros_packages/roslz4/package.xml
/opt/ros/share/roslz4
/opt/ros/share/roslz4/cmake
/opt/ros/share/roslz4/cmake/roslz4Config-version.cmake
/opt/ros/share/roslz4/cmake/roslz4Config.cmake
/opt/ros/share/roslz4/package.xml
/opt/ros/usr
/opt/ros/usr/lib64
/opt/ros/usr/lib64/python2.7
/opt/ros/usr/lib64/python2.7/site-packages
/opt/ros/usr/lib64/python2.7/site-packages/roslz4
/opt/ros/usr/lib64/python2.7/site-packages/roslz4-1.13.0-py2.7.egg-info
/opt/ros/usr/lib64/python2.7/site-packages/roslz4/__init__.py
/opt/ros/usr/lib64/python3.4
/opt/ros/usr/lib64/python3.4/site-packages
/opt/ros/usr/lib64/python3.4/site-packages/roslz4
/opt/ros/usr/lib64/python3.4/site-packages/roslz4-1.13.0-py3.4.egg-info
/opt/ros/usr/lib64/python3.4/site-packages/roslz4/__init__.py
/opt/ros/usr/lib64/python3.5
/opt/ros/usr/lib64/python3.5/site-packages
/opt/ros/usr/lib64/python3.5/site-packages/roslz4
/opt/ros/usr/lib64/python3.5/site-packages/roslz4-1.13.0-py3.5.egg-info
/opt/ros/usr/lib64/python3.5/site-packages/roslz4/__init__.py
/usr
/usr/lib
/usr/lib/debug
/usr/lib/debug/opt
/usr/lib/debug/opt/ros
/usr/lib/debug/opt/ros/lib64
/usr/lib/debug/opt/ros/lib64/libroslz4.so.debug
/usr/lib/debug/usr
/usr/lib/debug/usr/lib64
/usr/lib/debug/usr/lib64/python2.7
/usr/lib/debug/usr/lib64/python2.7/site-packages
/usr/lib/debug/usr/lib64/python2.7/site-packages/roslz4
/usr/lib/debug/usr/lib64/python2.7/site-packages/roslz4/_roslz4.so.debug
/usr/lib/debug/usr/lib64/python3.4
/usr/lib/debug/usr/lib64/python3.4/site-packages
/usr/lib/debug/usr/lib64/python3.4/site-packages/roslz4
/usr/lib/debug/usr/lib64/python3.4/site-packages/roslz4/_roslz4.so.debug
/usr/lib/debug/usr/lib64/python3.5
/usr/lib/debug/usr/lib64/python3.5/site-packages
/usr/lib/debug/usr/lib64/python3.5/site-packages/roslz4
/usr/lib/debug/usr/lib64/python3.5/site-packages/roslz4/_roslz4.so.debug
/usr/lib64
/usr/lib64/python2.7
/usr/lib64/python2.7/site-packages
/usr/lib64/python2.7/site-packages/roslz4
/usr/lib64/python2.7/site-packages/roslz4/_roslz4.so
/usr/lib64/python3.4
/usr/lib64/python3.4/site-packages
/usr/lib64/python3.4/site-packages/roslz4
/usr/lib64/python3.4/site-packages/roslz4/_roslz4.so
/usr/lib64/python3.5
/usr/lib64/python3.5/site-packages
/usr/lib64/python3.5/site-packages/roslz4
/usr/lib64/python3.5/site-packages/roslz4/_roslz4.so
```

this is when I build with

```
CATKIN_PREFIX_PATH=/opt/ros
CMAKE_PREFIX_PATH=/opt/ros
CMAKE_INSTALL_PREFIX=/opt/ros
ROS_ROOT=/opt/ros
DEST_SETUP_DIR=/opt/ros
PYTHON_INSTALL_DIR=/usr/lib64/pythonx.y/site-packages
```

if I change `PYTHON_INSTALL_DIR` to `PYTHON_INSTALL_DIR=/usr/lib64/pythonx.y/site-packages`
it became

```
/opt
/opt/ros
/opt/ros/include
/opt/ros/include/roslz4
/opt/ros/include/roslz4/lz4s.h
/opt/ros/lib64
/opt/ros/lib64/libroslz4.so
/opt/ros/lib64/pkgconfig
/opt/ros/lib64/pkgconfig/roslz4.pc
/opt/ros/opt
/opt/ros/opt/ros
/opt/ros/opt/ros/usr
/opt/ros/opt/ros/usr/lib64
/opt/ros/opt/ros/usr/lib64/python2.7
/opt/ros/opt/ros/usr/lib64/python2.7/site-packages
/opt/ros/opt/ros/usr/lib64/python2.7/site-packages/roslz4
/opt/ros/opt/ros/usr/lib64/python2.7/site-packages/roslz4-1.13.0-py2.7.egg-info
/opt/ros/opt/ros/usr/lib64/python2.7/site-packages/roslz4/__init__.py
/opt/ros/opt/ros/usr/lib64/python3.4
/opt/ros/opt/ros/usr/lib64/python3.4/site-packages
/opt/ros/opt/ros/usr/lib64/python3.4/site-packages/roslz4
/opt/ros/opt/ros/usr/lib64/python3.4/site-packages/roslz4-1.13.0-py3.4.egg-info
/opt/ros/opt/ros/usr/lib64/python3.4/site-packages/roslz4/__init__.py
/opt/ros/opt/ros/usr/lib64/python3.5
/opt/ros/opt/ros/usr/lib64/python3.5/site-packages
/opt/ros/opt/ros/usr/lib64/python3.5/site-packages/roslz4
/opt/ros/opt/ros/usr/lib64/python3.5/site-packages/roslz4-1.13.0-py3.5.egg-info
/opt/ros/opt/ros/usr/lib64/python3.5/site-packages/roslz4/__init__.py
/opt/ros/share
/opt/ros/share/ros_packages
/opt/ros/share/ros_packages/roslz4
/opt/ros/share/ros_packages/roslz4/package.xml
/opt/ros/share/roslz4
/opt/ros/share/roslz4/cmake
/opt/ros/share/roslz4/cmake/roslz4Config-version.cmake
/opt/ros/share/roslz4/cmake/roslz4Config.cmake
/opt/ros/share/roslz4/package.xml
/opt/ros/usr
/opt/ros/usr/lib64
/opt/ros/usr/lib64/python2.7
/opt/ros/usr/lib64/python2.7/site-packages
/opt/ros/usr/lib64/python2.7/site-packages/roslz4
/opt/ros/usr/lib64/python2.7/site-packages/roslz4/_roslz4.so
/opt/ros/usr/lib64/python3.4
/opt/ros/usr/lib64/python3.4/site-packages
/opt/ros/usr/lib64/python3.4/site-packages/roslz4
/opt/ros/usr/lib64/python3.4/site-packages/roslz4/_roslz4.so
/opt/ros/usr/lib64/python3.5
/opt/ros/usr/lib64/python3.5/site-packages
/opt/ros/usr/lib64/python3.5/site-packages/roslz4
/opt/ros/usr/lib64/python3.5/site-packages/roslz4/_roslz4.so
/usr
/usr/lib
/usr/lib/debug
/usr/lib/debug/opt
/usr/lib/debug/opt/ros
/usr/lib/debug/opt/ros/lib64
/usr/lib/debug/opt/ros/lib64/libroslz4.so.debug
/usr/lib/debug/opt/ros/usr
/usr/lib/debug/opt/ros/usr/lib64
/usr/lib/debug/opt/ros/usr/lib64/python2.7
/usr/lib/debug/opt/ros/usr/lib64/python2.7/site-packages
/usr/lib/debug/opt/ros/usr/lib64/python2.7/site-packages/roslz4
/usr/lib/debug/opt/ros/usr/lib64/python2.7/site-packages/roslz4/_roslz4.so.debug
/usr/lib/debug/opt/ros/usr/lib64/python3.4
/usr/lib/debug/opt/ros/usr/lib64/python3.4/site-packages
/usr/lib/debug/opt/ros/usr/lib64/python3.4/site-packages/roslz4
/usr/lib/debug/opt/ros/usr/lib64/python3.4/site-packages/roslz4/_roslz4.so.debug
/usr/lib/debug/opt/ros/usr/lib64/python3.5
/usr/lib/debug/opt/ros/usr/lib64/python3.5/site-packages
/usr/lib/debug/opt/ros/usr/lib64/python3.5/site-packages/roslz4
/usr/lib/debug/opt/ros/usr/lib64/python3.5/site-packages/roslz4/_roslz4.so.debug
```

So what variable combination I need to use?

Alessandro-Barbieri, 2017-07-06:
With `PYTHON_INSTALL_DIR=lib64/pythonx.y/site-packages` will install right.

dirk-thomas, 2017-07-06:
From your description it makes sense that if you pass a custom `PYTHON_INSTALL_DIR` that it needs to match the layout on your system. Can you please clarify what the original value of  `PYTHON_INSTALL_DIR` was (I assume it wasn't correct so that you started passing your own value).