ros/ros_comm #1093: Can't call ros::init twice

facontidavide, 2017-07-10:
Use case.

i have a GUI  application where by default the code will try to call ros::init with default values ro ROS_MASTER_URI.

If that fails, I have a dialog that ask for the correct URI. The code is not much different from the one here:
http://wiki.ros.org/qt_create

In pseudo-code, what I want to do is:
```

std::map<std::string,std::string> remappings;
remappings["__master"] = DEFAULT_URL;
remappings["__hostname"] = HOST_URL;
ros::init(remappings, NODE_NAME );

if ( ! ros::master::check() ) {
   remappings["__master"] = USER_PROVIDED_URL;
   ros::init(remappings, NODE_NAME );
}

```
I am positive that this kind of code used to work something like 1 year ago... but maybe I am wrong.
As I can see here, the variable g_initialized can not be set to false.

https://github.com/ros/ros_comm/blob/247459207e20c1da109fc306e58b84d15c4107bd/clients/roscpp/src/libros/init.cpp#L446

dirk-thomas, 2017-07-10:
Before calling `ros::init` a second time you will need to call `ros::shutdown` (https://github.com/ros/ros_comm/blob/df86805c5b155829f736fa581424d9fe439430b9/clients/roscpp/src/libros/init.cpp#L575) if you want to be able to reinitialize it with different parameters.

facontidavide, 2017-07-10:
Nope :)
I already tried that before submitting the issue.

**g_initialized** is not set to false by ros::shutdown, therefore lines between [446](https://github.com/ros/ros_comm/blob/247459207e20c1da109fc306e58b84d15c4107bd/clients/roscpp/src/libros/init.cpp#L446) and [464](https://github.com/ros/ros_comm/blob/247459207e20c1da109fc306e58b84d15c4107bd/clients/roscpp/src/libros/init.cpp#L464) are never executed twice.

dirk-thomas, 2017-07-10:
Closing as duplicate of #214. Please see the comments in that ticket why the global variable is currently not reset.

closed 2017-07-10 by dirk-thomas

facontidavide, 2017-07-10:
thanks, the fix in #214 works for me too