ros/ros_comm #1094: ros::this_node::getNamespace() ought to return sanitized namespace

lucasw, 2017-07-11:
```
  ros::NodeHandle nh("~blah");
  std::string ns = ros::this_node::getNamespace();
  std::ns);
  ROS_INFO_STREAM(nh.getNamespace());
```
This is fine
```
$ rosrun ns_test ns_test __ns:=foo
/foo
/foo/ns_test
```

This ought to print the same
```
rosrun ns_test ns_test __ns:=/foo
//foo
/foo/ns_test
```

```
rosrun ns_test ns_test __ns:=//foo//
///foo//
/foo/ns_test
```

Using kinetic currently

wjwwood, 2017-07-11:
Makes sense to me, though it might be someone is relying on this behavior (they want what came in over the CLI, not the sanitized version), but we can definitely change it for newer distributions and consider doing it anyways for existing distributions.

It's definitely worth making a pr against `lunar-devel` if you have the time.

lucasw, 2017-07-14:
I notice roslaunch 

```
<node name="ns_test"
    pkg="ns_test"
    type="ns_test"
    output="screen"
    ns="//foo//"/>
```

prints out

```
NODES
  //foo//
    ns_test (ns_test/ns_test)
...
process[/foo//ns_test-1]: started with pid [26179]
...
[/foo//ns_test-1] process has finished cleanly
```

The NODES part prints the namespace verbatim, while the process messages are removing a leading forward slash.

lucasw, 2017-07-14:
If there were to be a namespace cleaning test `test/test_roscpp/test/launch/namespace.xml` and `test/test_roscpp/test/src/namespaces.cpp` could be the basis for it.

(and build and test with 
```
catkin_make test_roscpp-namespaces-clean
rostest src/ros_comm/test/test_roscpp/test/launch/namespaces-clean.xml
```
)

dirk-thomas, 2017-07-19:
The argument `__ns` is not sanitized at the moment and the user is expected to pass a non-global namespace without a trailing slash.

I am not sure why you want to pass namespaces with leading and trailing namespaces in the first place but since the provided PR is straight forward it shouldn't be a problem to do the extra sanitization you proposed. Thanks for create the patch.

dirk-thomas, 2017-07-19:
Addressed by #1100.

closed 2017-07-21 by dirk-thomas

k-okada, 2017-07-25:
I ran into similar situation and still confused that on python , we'll get 
```
$ ROS_NAMESPACE=/hoge/ python -c 'import rospy; print(rospy.get_namespace())'
/hoge/
$ ROS_NAMESPACE=/hoge python -c 'import rospy; print(rospy.get_namespace())'
/hoge/
$ ROS_NAMESPACE=hoge python -c 'import rospy; print(rospy.get_namespace())'
/hoge/
$ ROS_NAMESPACE=hoge/ python -c 'import rospy; print(rospy.get_namespace())'
/hoge/
```
on rospy, but on roscpp (after applied this patch), we'll get

```
$ ROS_NAMESPACE=/hoge/ ../../devel/lib/test_namespace/test_namespace
[ INFO] [1501025799.775532508]: namespace -> /hoge
$ ROS_NAMESPACE=/hoge ../../devel/lib/test_namespace/test_namespace
[ INFO] [1501025804.109156084]: namespace -> /hoge
$ ROS_NAMESPACE=hoge ../../devel/lib/test_namespace/test_namespace
[ INFO] [1501025806.913568473]: namespace -> /hoge
$ ROS_NAMESPACE=hoge/ ../../devel/lib/test_namespace/test_namespace
[ INFO] [1501025810.195600461]: namespace -> /hoge
```
the source code is
```
$ cat src/test_namespace.cpp
#include <ros/ros.h>

int main(int argc, char **argv) {
  ros::init(argc, argv, "test_namespace");
  ROS_INFO_STREAM("namespace -> " <<  ros::this_node::getNamespace());
}
```

so question is, do we have tailing '/' on namespace or not.