ros/ros_comm #1106: Remote Launch with log4cxx display on console

littlejohnyang, 2017-07-20:
I have written a launch file to remotely start nodes from another machine. However I won't be able to display the debug output on the console if I use this remote launch file rather than having a ssh script into the machine. (I can only see "ERROR" messages but not other debug level.

I know I can normally configure the logging output option by setting:
`<env name="ROSCONSOLE_CONFIG_FILE" value="~/catkin_ws/mypackage/config/custom_rosconsole.config"/>	`
However, this returns saying this config file being not found (even though this file is definitely there on both machine).

dirk-thomas, 2017-07-21:
The behavior you are expecting is not supported by `roslaunch` atm. `roslaunch` only prints `stderr` from remote processes (https://github.com/ros/ros_comm/blob/fad91041fa322a21add175cac4872ad75f9219da/tools/roslaunch/src/roslaunch/remoteprocess.py#L259-L270). `stdout` is not being printed anywhere (https://github.com/ros/ros_comm/blob/fad91041fa322a21add175cac4872ad75f9219da/tools/roslaunch/src/roslaunch/remoteprocess.py#L280-L288).

You might want to subscribe to the `rosout` topic to receive the log message from all nodes instead.

peci1, 2021-09-09:
We implemented this feature as a monkey-patch of roslaunch python library. You'll find it here: https://github.com/ctu-vras/scoped_roslaunch . The package was originally created to overcome another roslaunch issue, so don't get confused by the group/scope-related stuff. However, if you're serious about using remote machine tags, it also comes helpful.

I tested the patched library and it works without issues. As a bonus, you get correct coloring of warning messages (yellow, not red), smaller read buffer (so you see the messages earlier) and line buffering (so that remote lines are not broken in the middle just on the boundary of read sizes).

This is how it looks on our 3-computer robot:

[image]
.

peci1, 2021-09-09:
If anybody else could give these changes a test and report here, maybe we could port them to upstream roslaunch. The change doesn't break any API or ABI (it just changes what users see in the console, which might be confusing; but I would say confusing is the current state).

closed 2025-05-31 by sloretz