ros/ros_comm #1108: Why does rosbag_storage add "topic" to the connection header?

jtbandes, 2017-07-20:
[I posted this [on answers.ros.org](http://answers.ros.org/question/266967/why-does-rosbag_storage-add-topic-to-the-connection-header/) but it was suggested that I also ask here...]

rosbag_storage makes a copy of the connection header and adds the `"topic"` to it: https://github.com/ros/ros_comm/blob/indigo-devel/tools/rosbag_storage/include/rosbag/bag.h#L502-L503

This was apparently added to mitigate an issue introduced by https://github.com/ros/ros_comm/commit/5362f2cf8a6282baf569974a3b13af3723ebb92d

However, as far as I can tell the topic should already be in the header (and experimentally, when I subscribe with a `MessageEvent` callback, `event.getConnectionHeader()["topic"]` does have the topic name): https://github.com/ros/ros_comm/blame/indigo-devel/clients/roscpp/src/libros/intraprocess_publisher_link.cpp#L72

Am I missing something? Why was it necessary to add the topic separately?

Unfortunately, the old relevant trac issues 3727 (motivation for 5362f2cf8a6282baf569974a3b13af3723ebb92d) / 3755 (issue apparently arising as a result of that change) are gone:  
http://lists.ros.org/pipermail/ros-users/2011-December/065317.html  
http://lists.ros.org/pipermail/ros-users/2011-October/065018.html

Thanks!

dirk-thomas, 2017-07-21:
Since you already raised your question on answers.ros.org I will close this as a duplicate.

I don't have an answer for your question. I can only speculate that the `topic` field is (or at least was) not set in all cases. With the referenced tickets being unavailable I don't any more insight.

Maybe @tfoote can recover the information from code.ros.org/trac/ros/ticket/3755. He should have a full backup of that system somewhere.

closed 2017-07-21 by dirk-thomas

jtbandes, 2017-07-21:
Or perhaps @gerkey would have some insight as the author of https://github.com/ros/ros_comm/commit/5362f2cf8a6282baf569974a3b13af3723ebb92d?

gerkey, 2017-07-26:
Wow, that's from the way-back machine.

I stared at the code for a while and unfortunately cannot page in any relevant memories of that change.

My hypothesis is that there is/was a situation in which you can end up in `Bag::doWrite()` with the `connection_header` that you're passed not containing the `topic`. Alternatively, perhaps I added an unnecessary copy there, which is also possible.

Is that copy causing a problem? You could try removing it and see if the test suite still passes.

jtbandes, 2017-07-26:
It's not causing a problem, but I would just like to see if I'm missing something in understanding how this works :)

I forgot to link to this earlier, but it looks like the copy of `topic` was introduced in https://github.com/ros/ros_comm/commit/0120fd0da0d17a1aab7e0e364557eb3887f4f432 Based on the other stuff in this commit, and the comment, it looks like an attempt to distinguish publishers that are identical except for the same topic. However since in my experience the `connection_header` already contains the topic, it doesn't matter. Maybe there's a special case when there is already a connection between that publisher/subscriber pair. I'll see if I can get the tests up and running...

jtbandes, 2017-07-26:
ok, tried removing `connection_header_copy["topic"] = topic;` and running the tests and it still seems to work... yet it seems pretty clear this code (and this test) was added for a good reason at some point in the past... 🤷‍♂️  😕

gerkey, 2017-07-26:
Can you further try removing the use of `connection_header_copy` altogether? If the tests still pass, then we could consider making that change.

dirk-thomas, 2017-07-26:
While the tests cover a good amount of cases there are so many more cases not covered that any unnecessary change implies a chance for regression. If there is not reason to remove / alter the current code I would rather not touch it.