ros/ros_comm #1111: Why are values greater than 1 raising error for "if" and "unless" statements?

cagbal, 2017-07-26:
Hi All, 

In the documentation: 

> All tags support if and unless attributes, which include or exclude a tag based on the evaluation of a value. "1" and "true" are considered true values. "0" and "false" are considered false values. Other values will error.

So, if I write:
`<group unless="2">
   </group>`

It says: 

> 2 is not a 'bool' type

Normally, "0" should be False and all other numbers should be True, right? This just limits the usage, I think. What is the reason of this usage?

dirk-thomas, 2017-07-26:
Just for reference: the attribute is being read [here](https://github.com/ros/ros_comm/blob/6814ebb5df98c34c711dce854bf6304ae3e6b616/tools/roslaunch/src/roslaunch/xmlloader.py#L81) which than uses the `convert_value` function for [bool](https://github.com/ros/ros_comm/blob/6814ebb5df98c34c711dce854bf6304ae3e6b616/tools/roslaunch/src/roslaunch/loader.py#L91-L97).

> Normally, "0" should be False and all other numbers should be True

While this is certainly one way of handling the conversion it is not the only one. The current implementation choice dates back to the very early days of ROS (+7 yrs ago) so I don't know the rational for the decision.

If you have a strong opinion about it please consider proposing a patch to relax the current behavior. Also make sure to update corresponding tests and be prepared to update existing documentation once the PR has been accepted.

cagbal, 2017-07-27:
Oh, yes actually, it is a one way to do this :) Let me discuss it with others and then I may propose a patch. Thanks for your kind response. I am not closing the issue for now; so that, other people also can comment on this.

closed 2018-03-12 by cagbal