ros/ros_comm #1143: support steady time on OS X

dirk-thomas, 2017-08-14:
Currently it falls back to wall time.

flixr, 2017-08-14:
Just to be pedantic and to avoid confusion: I think `SteadyTime` (from the rostime package) should work on OSX, but not the `SteadyTimer`... at least #1142 is about the `SteadyTimer`

flixr, 2017-08-14:
This will probably become a bit more complicated... it will need fixing of [boost pthread condition variable](https://github.com/ros/ros_comm/blob/a46a0fe7f69cdd702ff15aaca3c2a1260a9be98f/clients/roscpp/include/boost_161_pthread_condition_variable_fwd.h#L45)

The things @clalancette mentioned in #1142 are really about getting the steady/monotonic time (as @wjwwood already seems to have fixed for `SteadyTime` with https://github.com/ros/roscpp_core/pull/59) and not about setting the clock for a boost condition variable as what is needed here...