ros/ros_comm #1145: rospack.h and bag.h cannot both be included in the same compilation unit

jstnhuang, 2017-08-18:
A simple program like:
```cpp
#include "rosbag/bag.h"
#include "rospack/rospack.h"

int main(int argc, char** argv) { return 0; }
```

Fails to compile with this error:
```
[100%] Building CXX object rospack_test/CMakeFiles/rospack_test_node.dir/src/main.cpp.o
In file included from /home/jstn/asdf_ws/src/rospack_test/src/main.cpp:2:0:
/opt/ros/indigo/include/rospack/rospack.h:539:1: error: expected identifier before string constant
                  bool append_errno = false);
 ^
/opt/ros/indigo/include/rospack/rospack.h:539:1: error: expected ‘,’ or ‘...’ before string constant
In file included from /opt/ros/indigo/include/rosbag/bag.h:63:0,
                 from /home/jstn/asdf_ws/src/rospack_test/src/main.cpp:1:
/opt/ros/indigo/include/rospack/rospack.h:538:10: error: invalid use of ‘::’
     void logWarn(const std::string& msg,
          ^
In file included from /home/jstn/asdf_ws/src/rospack_test/src/main.cpp:2:0:
/opt/ros/indigo/include/rospack/rospack.h:547:1: error: expected identifier before string constant
                   bool append_errno = false);
 ^
/opt/ros/indigo/include/rospack/rospack.h:547:1: error: expected ‘,’ or ‘...’ before string constant
In file included from /opt/ros/indigo/include/rosbag/bag.h:63:0,
                 from /home/jstn/asdf_ws/src/rospack_test/src/main.cpp:1:
/opt/ros/indigo/include/rospack/rospack.h:546:10: error: invalid use of ‘::’
     void logError(const std::string& msg,
          ^
make[2]: *** [rospack_test/CMakeFiles/rospack_test_node.dir/src/main.cpp.o] Error 1
make[1]: *** [rospack_test/CMakeFiles/rospack_test_node.dir/all] Error 2
make: *** [all] Error 2
```

However, I was able to separate out the rospack functionality into a separate library, and link it with my main executable (which used rosbag) and successfully compile my code.

Tested on Ubuntu 14.04 with ROS Indigo, using catkin and catkin_tools.

clalancette, 2017-08-21:
I see what the problem is.

In `rospack/rospack.h`, the Rosstackage class has two public methods called `logWarn` and `logError`.  Unfortunately, these are the names of macros defined in `console_bridge/console.h`, which gets pulled in recursively by `rosbag/bag.h`.  Thus, when you include `rosbag/bag.h` first, it defines those macros and then replaces the code in `rospack/rospack.h`.  These macros aren't really viable at that point, so it causes bad syntax (essentially).

I'm not quite sure what the fix is, unfortunately.  The easiest thing to do would be to rename these two methods in Rosstackage, but that will probably break a downstream user.  Another thing we could do is use `#pragma push_macro("logWarn")` to undefine the macro for the length of the header file, then `#pragma pop_macro("logWarn")` to redefine it at that end.  It is a non-standard pragma extension, but it looks like gcc, clang, and MSVC all support it.  @dirk-thomas Any thoughts about this latter approach, or other ideas for fixing this?

clalancette, 2017-08-21:
(technically this issue belongs over at rospack; I'll try to move it over there in a bit)

dirk-thomas, 2017-08-21:
The problem is that `rosbag/bag.h` include `console_bridge/console.h` which defines these macros. While they have been deprecated for a long time exactly for this kind of problem they haven't been removed yet.

Please see #1149 for the proposed fix. It would be great if you could try the patch and comment here if it solves your problem. Thanks.

> (technically this issue belongs over at rospack; I'll try to move it over there in a bit)

`rospack` neither includes console_bridge nor uses it anywhere. It only happens to have methods which have the same name as these deprecated macros from `console_bridge`. So `rospack` shouldn't be concerned with this.

jstnhuang, 2017-08-21:
I can't test it because I'm using Indigo. If I try cherry-picking the commit to ros_comm@indigo-devel, it doesn't compile because the Indigo version of `console_bridge` doesn't define CONSOLE_BRIDGE_logError, etc.

dirk-thomas, 2017-08-21:
@clalancette Can you try the patch and confirm that it addresses the problem?

clalancette, 2017-08-22:
@dirk-thomas Yes, will do and report back.

PR #1149 (2017-08-28): use not deprecated console_bridge macros and undefine the deprecated ones
Fixes #1145.

This patch avoids the side effect of defining these macro which are often subject to collisions in a ROS header.
closed 2017-08-28 by dirk-thomas