ros/ros_comm #1148: Configuration across IPV6 - Kinetic

richard-a-mck, 2017-08-21:
Hi there. I'm having some issues with configuration of ROS across multiple machines using an ad-hoc network on a PC with Ubuntu 16, ROS Kinetic and a Raspberry Pi with Ubuntu 16 and ROS Kinetic. 
Someone on the ROS answers board suggested I raise an issue here. Here's the [link to my original post](https://answers.ros.org/question/269182/correct-ros_master_uri-format-for-ipv6/).
Of interest, on my PC if I set the master URI to 

**ROS_MASTER_URI=http://[fe80::ba27:ebff:feb8:fee8%wlxf4f26d11d066]:11311** 

and run _roswtf_ I get the following output:

```
Loaded plugin tf.tfwtf
No package or stack in context
Traceback (most recent call last):
  File "/opt/ros/kinetic/bin/roswtf", line 35, in <module>
    roswtf.roswtf_main()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roswtf/__init__.py", line 89, in roswtf_main
    _roswtf_main()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roswtf/__init__.py", line 172, in _roswtf_main
    wtf_check_environment(ctx)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roswtf/environment.py", line 219, in wtf_check_environment
    error_rule(r, r[0](ctx), ctx)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roswtf/rules.py", line 100, in error_rule
    _check_rule(rule, ret, ctx, ctx.errors, WtfError)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roswtf/rules.py", line 71, in _check_rule
    ctx_list.append(level(f_msg%d + ret%d, f_msg, ret))
ValueError: unsupported format character 'w' (0x77) at index 51
```

If I change my master_uri to the following, 

**ROS_MASTER_URI=http://[fe80::ba27:ebff:feb8:fee8%%wlxf4f26d11d066]:11311**
 
(note the double %), I get the following roswtf output:

```
Loaded plugin tf.tfwtf
No package or stack in context
================================================================================
Static checks summary:

Found 1 error(s).

ERROR ROS_MASTER_URI [http://[fe80::ba27:ebff:feb8:fee8%%wlxf4f26d11d066]:11311] is not a valid URL: invalid address string [[fe80::ba27:ebff:feb8:fee8%wlxf4f26d11d066]:11311]
================================================================================

ROS Master does not appear to be running.
Online graph checks will not be run.
ROS_MASTER_URI is [http://[fe80::ba27:ebff:feb8:fee8%%wlxf4f26d11d066]:11311]
```

I can successfully ping6 the address of the RPI (fe80... etc) and vice versa from the RPI. I can also ssh between the two devices without issue.

Is this a genuine issue or am I just misconfiguring something?

richard-a-mck, 2017-08-21:
Also worth noting I have added _export ROS_IPV6=on_ to ~/.bashrc on both systems.

dirk-thomas, 2017-08-21:
Sorry just noticed the link to the existing question... Will reopen this again and try to answer...

dirk-thomas, 2017-08-21:
In many locations the string containing the *uri* will be used and formatted using the `%` operator. Since classically it only contained IPv4 addresses it didn't contain `%` characters. In your case it does and when being used with the `%` operator it is being misinterpreted as `%w` which then fails.

This will likely need some substitutions of the single `%` with a double `%%` (similar to #585). It would be great if your could look into where that is necessary and propose a pull request to make it work for your use case.

richard-a-mck, 2017-08-21:
After some quick investigation, it looks like the fix in #585 was only applied to Indigo. My network.py file still has the original command. I'll try submit a PR to change this for Kinetic

dirk-thomas, 2017-09-14:
> After some quick investigation, it looks like the fix in #585 was only applied to Indigo.

The patch from #598 is available in all current branches - also in the latest `lunar-devel`.

> My network.py has the following which should in theory fix this problem:

The snipped you posted is exactly the same as the code currently on the `lunar-devel` branch: https://github.com/ros/ros_comm/blob/67633983de134d7e4c55c4d28956aa603eb42099/utilities/roswtf/src/roswtf/network.py#L47-L60

Therefore I am not sure what you are proposing and if the problem still exists for you.

richard-a-mck, 2017-09-15:
The section I posted is supposed to be the code that fixes the IPv6 bug. It is also the code that is in-use on my system, where the IPv6 bug is present, suggesting that the fix doesn't work, at least in my case.
I've worked around it by reverting to IPv4 for this project. It would be good to test further with IPv6 addresses to figure out whether this was an issue unique to me or not. In saying that, I don't have sufficient understanding of python or the ROS network stack to resolve this myself.

smithan7, 2019-03-21:
I am having the same issue with IPV6, any update on this? Has anyone had success with IPV6 and can point to a resource?

closed 2025-05-31 by sloretz