ros/ros_comm #1153: [rosrun] private param parsing as yaml

guihomework, 2017-08-24:
in kinetic, it seems to be impossible to enforce to pass a string to a private parameter in rosrun

if myapp wants to get a string like so 
```
std::string myvariable;
nh_private.getParam("myparam", myvariable);
```
this will return false when parsing / casting to a string if a number is given even if put in quotes

```
rosrun mypkg myapp _myparam:=123
rosrun mypkg myapp _myparam:="123"
rosrun mypkg myapp _myparam:='123'
```
I though I could force the param the same way as in service calls
```
rosservice call /my_service '!!str true'
```

but apparently rosrun does not parse private parameters as yaml
```
rosrun mypkg myapp _myparam:='!!str 123'
```
gets myvariable set to '''!!str 123'''

So the only solution I found is to handle the XmlRpc myself and convert INTs into a string if ```getType()==XmlRpc::XmlRpcValue::TypeInt```

evenator, 2017-08-25:
This is actually a roscpp bug, not a rosrun bug. The code in question is in `param::init()` in [clients/roscpp/src/libros/param.cpp](https://github.com/ros/ros_comm/tree/lunar-devel/clients/roscpp/src/libros/param.cpp)

jdlangs, 2017-10-27:
It might be useful to have an exception thrown if the parameter function argument type doesn't match the value existing in the server.

More insidious than the `getParam` case are the `NodeHandle::param` functions, which silently give you the default value. It would have just saved me a good chunk of time if I had received an immediate error that the parameter type is mismatched with the template argument.

closed 2025-05-31 by sloretz