ros/ros_comm #1161: Bloom release archive doesn't match github version tag?

de-vri-es, 2017-09-11:
I was building roscpp from the 1.13.2 bloom release archive, and it didn't compile due to a missing include which was fixed somewhere in July. Then I compared it with the 1.13.2 tag here, and they are not the same files.

I don't know how it happened. Maybe something went wrong while making the bloom release? Either way, it seems like a big problem.

The bloom release[1] has this:

`ros_comm-release-release-kinetic-roscpp/src/libros/transport/transport_udp.cpp`:
```
#include <fcntl.h>
#if defined(__APPLE__)
  // For readv() and writev()
  #include <sys/types.h>
  #include <sys/uio.h>
  #include <unistd.h>
#elif defined(__ANDROID__)
  // For readv() and writev() on ANDROID
  #include <sys/uio.h>
#endif
```

The github release[2] has this:

`ros_comm-1.13.2/clients/roscpp/src/libros/transport/transport_udp.cpp`:
```
#include <fcntl.h>
#if defined(__APPLE__)
  // For readv() and writev()
  #include <sys/types.h>
  #include <sys/uio.h>
  #include <unistd.h>
#elif defined(__ANDROID__)
  // For readv() and writev() on ANDROID
  #include <sys/uio.h>
#elif defined(_POSIX_VERSION)
  // For readv() and writev()
  #include <sys/uio.h>
#endif
```

[1] https://github.com/ros-gbp/ros_comm-release/archive/release/kinetic/roscpp/1.13.2.tar.gz
[2] https://github.com/ros/ros_comm/archive/1.13.2.tar.gz

de-vri-es, 2017-09-11:
Oh, is 1.13.2 even released for kinetic?

wjwwood, 2017-09-11:
I don't know why `1.13.2` exists for kinetic, but last version I see released for kinetic is `1.12.7`:

-
 https://github.com/ros/rosdistro/blob/b3bc5cf5476ff810047f6501a221ff448d62ab28/kinetic/distribution.yaml#L6930
- https://github.com/ros-gbp/ros_comm-release#ros_comm-kinetic---1127-0

@dirk-thomas may know. I don't think this is an issue with bloom, however, and I do not believe the `1.13.2` version you're looking at was released into Kinetic, as you pointed out.

closed 2017-09-11 by wjwwood

de-vri-es, 2017-09-11:
Looking at the wget log, github is redirecting to `https://codeload.github.com/ros-gbp/ros_comm-release/tar.gz/release/kinetic/roscpp`.

So it seems to be githubs fault for blanket redirecting non-existing archives to something else. If there is something that can be done to disable that, it would be an improvement. A 404 error is much more useful that getting the wrong version unknowingly.

wjwwood, 2017-09-11:
That does suck, but I don't see any settings like that in the repository admin page. Honestly it sounds like a bug to me. You might want to report it to GitHub.