ros/ros_comm #1167: How to have a specific node survive when a required node dies from a roslaunch?

WesleyYue, 2017-09-19:
Let's say I have nodes A, B, C ... Z. I want for nodes C to Z to die when A dies, but for B to continue executing and ignore that A died. I know I can use the `required="true"` argument on A in the roslaunch file to kill other nodes when it dies, but is there a way to tell roslaunch to not kill B?

dirk-thomas, 2017-09-20:
I am not sure if you have seen the [ROS support guidelines](http://wiki.ros.org/Support) but we kindly ask to raise questions like this on [answers.ros.org](http://answers.ros.org/) instead.

The rational for asking all kind of questions in a single place is:
* There are many more people reading questions there so your chances on getting an answer and in a timely manner are much higher.
* In the future other users will search there for similar problems and can find your question and the potential answers.
* The issue tracker in this repo is used to track bugs, feature requests, etc. But your question is not such an action item.

Therefore I will close this ticket here for now. If you have asked your question on answers.ros.org please feel free to add a link to the question to this ticket so that future readers can find the related question.

closed 2017-09-20 by dirk-thomas

WesleyYue, 2017-09-21:
https://answers.ros.org/question/271193/how-to-have-a-specific-node-survive-when-a-required-node-dies-from-a-roslaunch/

Since this is not currently possible, can this issue be reopened as a feature request?

tfoote, 2017-09-22:
If you would like to open a feature request, you can. This ticket does not really have enough information to understand what use cases you expect it to cover and which ones not to cover. In general you seem to want to have full control of shutdown behavior for each node relative to the potential states of the other nodes. This is a level of complexity that is explicitly avoided in the design of roslaunch. And since there's a simple solution of just launching B separately adding the complexity to roslaunch and it's syntax would be hard to justify.