ros/ros_comm #1169: CallbackQueue::clear() occasionally breaks subscriber

schallerr, 2017-09-22:
Hi there, 
I'm having issues with `CallbackQueue::clear()`. I have some `Subscriber`s associated with a callback queue. Occasionally, when `CallbackQueue::clear` is called on that queue, one or multiple of the `Subscriber`s stop working, meaning that no callbacks are triggered any more.

Looking at the implementation of `Subscription` it seems that this issue is related to [these lines](https://github.com/ros/ros_comm/blob/1e67842d7624205d94ed697c830c455d3f099db1/clients/roscpp/src/libros/subscription.cpp#L649-L658):
```
info->subscription_queue_->push(info->helper_, deserializer, info->has_tracked_object_, info->tracked_object_, nonconst_need_copy, receipt_time, &was_full);

if (was_full)
{
   ++drops;
}
else
{
   info->callback_queue_->addCallback(info->subscription_queue_, (uint64_t)info.get());
}
```

If the subscription queue is not empty when `CallbackQueue::clear` is called, the associated callbacks are removed but the items in the subscription queue aren't. Thus, in above code snippet, if the subscription queue is full, `SubscriptionQueue::push` sets `was_full` and no callback will be added to the callback queue. Therefore, the subscription queue stays full forever.

dirk-thomas, 2017-09-25:
Can you please provide [reproducible](http://sscce.org/) example. It would also be great if you could provide a PR since you already seem to have a good idea how to change the code to make it work.

schallerr, 2017-09-27:
I created an example here: https://github.com/schallerr/callbackqueue_clear_example
Unfortunately I don't really know how to solve the problem, I think there needs to be a way to ensure that items in the subscription queue are removed when the callback queue is cleared.

mikepurvis, 2017-09-27:
Thank you for providing a detailed example. I agree that this looks like a bug— my interpretation is that the logic assumes that anything in the subscription queue has already had a callback triggered for it. I agree with your assessment that when `CallbackQueue::clear()` is called, the associated subscription queue should _also_ be flushed.

The tricky thing here is that the `CallbackQueue` instance doesn't actually have any visibility to any associated subscriptions, since callbacks could be anything (Timers, parameters, whatever). So the key I think would be to set up a destructor on the queue items so that when they are popped and destroyed, they also correctly pop any items they would have grabbed from other queues.

All that said, I'm having trouble seeing how this would actually be implemented— it's `CallbackInfo` objects that go into the queues, but it seems that for subscriptions, there's no `SubscriptionCallbackInfo` object, rather it's the entire `SubscriptionQueue` that gets a pointer to itself queued up (here's it [popping a message in getting ready to pass it to the user callback](https://github.com/ros/ros_comm/blob/lunar-edge/clients/roscpp/src/libros/subscription_queue.cpp#L146)). So there may not be a destructor to hinge on, but perhaps a new optional polymorphic `CallbackInterface::cancel()` hook could be added, which offers an opportunity for the callback to do this kind of cleanup operation.