ros/ros_comm #1171: Inconsistent exit codes of tools in error cases

luator, 2017-09-25:
I just noticed that the console tools (`rosnode`, `rostopic`, ...) are inconsistent in their return code in case of errors.

The case I tested was running them without a running roscore (using kinetic):
* `rostopic list` --> exit code 1
* `rosservice list` --> exit code 2
* `rosnode list` --> exit code 0
* `rosparam list` --> exit code 1

Especially the exit code of `rosnode` is problematic if you expect to get an error code like with the other tools.

dirk-thomas, 2017-09-25:
All tools should indeed return a non-zero exit code in case of errors. Some have specific codes to indicate the exact error though.

Please consider providing a pull request to fix the error code of `rosnode list`.

luator, 2017-09-26:
Alright, I will try to find some time for it in the next days.

PR #1178 (2017-10-05): rosnode: Return exit code 1 if there is an error.
Return with exit code 1 if there is an error in the rosnode tool (e.g. because roscore is not running).
This resolves #1171.

I tested manually by calling `rosnode list` with both roscore running/not running and the result was good. I also ran `rostest rosnode rosnode.test` and `nosetests test_rosnode_command_offline.py`, both passed.
Please let me know in case I should do more testing (I do not know how to provoke other errors than roscore not running, though).

Since I only have a running workstation with Kinetic at the moment, I did the PR f [... cut]
closed 2017-10-05 by dirk-thomas