ros/ros_comm #1175: catkin_make tries to run a python2 script with python3

jerry73204, 2017-10-01:
## Environment
1. ArchLinux
2. `ros-kinetic-robot` package and  its all depends are install from AUR
3. `/usr/bin/python` is symlinked to python3, `/usr/bin/python2` is installed from separate package
4. `python2-catkin-pkg` is installed but `python-catkin-pkg` (py3 version) isn't due to file conflicts on /usr/bin/catkin_* files

## Description
By running `/opt/ros/kinetic/bin/catkin_make`:
```
####
#### Running command: "cmake /home/jerry73204/folder/src -DCATKIN_DEVEL_PREFIX=/home/jerry73204/folder/devel -DCMAKE_INSTALL_PREFIX=/home/jerry73204/folder/install -G Unix Makefiles" in "/home/jerry73204/folder/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/jerry73204/folder/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/lunar
-- This workspace overlays: /opt/ros/lunar
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using default Python package layout
-- Using empy: /usr/lib/python3.6/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/jerry73204/folder/build/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-3.6
ImportError: "from catkin_pkg.package import parse_package" failed: No module named 'catkin_pkg'
Make sure that you have installed "catkin_pkg", it is up to date and on the PYTHONPATH.
CMake Error at /opt/ros/lunar/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/usr/bin/python
  "/opt/ros/lunar/share/catkin/cmake/parse_package_xml.py"
  "/opt/ros/lunar/share/catkin/cmake/../package.xml"
  "/home/jerry73204/folder/build/catkin/catkin_generated/version/package.cmake")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/lunar/share/catkin/cmake/catkin_package_xml.cmake:74 (safe_execute_process)
  /opt/ros/lunar/share/catkin/cmake/all.cmake:151 (_catkin_package_xml)
  /opt/ros/lunar/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:52 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/jerry73204/folder/build/CMakeFiles/CMakeOutput.log".
See also "/home/jerry73204/folder/build/CMakeFiles/CMakeError.log".
```

What went wrong are the lines following. `/usr/bin/python` is Python 3 interpreter, while `/opt/ros/lunar/share/catkin/cmake/parse_package_xml.py` is apparently a Python 2 script assumed by its shebang.
```
  execute_process(/usr/bin/python
  "/opt/ros/lunar/share/catkin/cmake/parse_package_xml.py"
  "/opt/ros/lunar/share/catkin/cmake/../package.xml"
  "/home/jerry73204/folder/build/catkin/catkin_generated/version/package.cmake")
  returned error code 1
```

It seems that it didn't identify the the py2 interpreter as `/usr/bin/python2`, and incorrectly assumed `/usr/bin/python` to be a py2 interpreter.

Installing catkin_pkg in Python 3 maybe a workaround. Unfortunately, `python2-catkin-pkg` conflicts with `python-catkin-pkg`, and `python2-catkin-pkg` is required by `ros-kinetic-robot`.

tfoote, 2017-10-02:
If you've set python3 to be the default it seems to me that using python2-catkin-pkg is not going to work since it won't be on your PYTHONPATH.

Note that we've recently split the package into two parts to resolve the side by side installation issue with python2 and python3 scripts but the module is resolvable. https://github.com/ros-infrastructure/catkin_pkg/pull/157

I'd suggest ticketing the Arch packaging to switch their dependency to be on the module instead of the executable. It may also require updating the Arch packaging of catkin_pkg to split the package output too. That should resolve your side by side installation issue.

jerry73204, 2017-10-03:
I see there are relevant discussions on [AUR](https://aur.archlinux.org/packages/python-catkin_pkg/). As a workaround, I use `virtualenv` to resolve this issue.
`virtualenv2 --system-site-packages <env_dir>`

dirk-thomas, 2017-10-04:
As mentioned by @tfoote you need `catkin_pkg` installed for Python 3. Since there is nothing to be done here in `catkin` I will close this issue. Please feel free to continue commenting.

closed 2017-10-04 by dirk-thomas