ros/ros_comm #1185: ros python multi thread

yangshuo0619, 2017-10-12:
I'm doing a python project, there is a need to open a thread in the ROS service request, but I am looking for a long time in the ROS wiki, or did not solve, if you can give me some advice, thank you very much.

dirk-thomas, 2017-10-12:
I am not sure if you have seen the [ROS support guidelines](http://wiki.ros.org/Support) but we kindly ask to raise questions like this on [answers.ros.org](http://answers.ros.org/) instead.

The rational for asking all kind of questions in a single place is:
* There are many more people reading questions there so your chances on getting an answer and in a timely manner are much higher.
* In the future other users will search there for similar problems and can find your question and the potential answers.
* The issue tracker in this repo is used to track bugs, feature requests, etc. But your question is not such an action item.

Therefore I will close this ticket here for now. If you have asked your question on answers.ros.org please feel free to add a link to the question to this ticket so that future readers can find the related question.

closed 2017-10-12 by dirk-thomas

yangshuo0619, 2017-10-13:
I'm sorry, thanks