ros/ros_comm #1194: More "agile" demux?

peci1, 2017-10-19:
Hi, we've found the implementation of demux is quite slow when switching topics. Or at least the topic connections that need to be re-created during every switch are slow. The result is the switching happens quickly, but any subscribers start receiving messages a few seconds later (up to 15 seconds in my measurement).

Steps to reproduce:

    rosrun topic_tools demux a b c

    rostopic echo b

    rostopic echo c

    rostopic pub  -r 10 /a stamped_msgs/Int32 "header:
      seq: 0
      stamp:
        secs: 0
        nsecs: 0
      frame_id: ''
    data: 0"    

    rosservice call /demux/select b # (or c)

Try calling the `select` service a few times, and you'll see the long delay between a topic gets selected and the `rostopic`s start printing the new values.

What about giving `demux` a parameter that would tell it to keep all the publishers running all the time? Or even throwing away the old behavior and having the new one for all? I don't see any drawbacks, since only one of the publishers is doing some real work at a time...

dirk-thomas, 2017-10-20:
Sounds like a good improvement to me. Please consider to provide a pull request for this enhancement.

peci1, 2017-10-20:
Here we go...

dirk-thomas, 2018-02-03:
Addressed by #1196.

closed 2018-02-03 by dirk-thomas