ros/ros_comm #1201: roslaunch tests fail due to missing rospy/test_nodes/talker.py when rospy is installed as binary package

racko, 2017-10-24:
I thought it was a good idea to actually run the tests for the ros packages I maintain for arch linux. The following test failure occurred:
```
======================================================================
ERROR: test_create_node_process (unit.test_nodeprocess.TestNodeprocess)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "path/to/arch-ros-stacks/lunar/roslaunch/src/ros_comm-release-release-lunar-roslaunch-1.13.2-0/test/unit/test_nodeprocess.py", line 128, in test_create_node_process
    p = create_node_process(run_id, n, master_uri)
  File "/opt/ros/lunar/lib/python2.7/site-packages/roslaunch/nodeprocess.py", line 139, in create_node_process
    args = create_local_process_args(node, machine)
  File "/opt/ros/lunar/lib/python2.7/site-packages/roslaunch/node_args.py", line 256, in create_local_process_args
    raise NodeParamsException("can't locate node [%s] in package [%s]"%(node.type, node.package))
NodeParamsException: can't locate node [talker.py] in package [rospy]
```

The cause is rospy not installing it's test nodes (understandably).

How should this be dealt with?
* Should rospy just install the test_nodes?
* Should the test_nodes be moved to another package that can be specified as a test_depend? (My favourite)
* Should the test be disabled?
* Is my assumption wrong that the tests should run with dependencies installed as binary packages?

mikepurvis, 2017-10-24:
Many of the tests assume that they are in the devel space environment. There are many instances of packages not installing their test assets.

dirk-thomas, 2017-10-24:
@racko Can you please provide more information about your package and what you are doing in the test which will hopefully help to understand why it fails (beyond `rospy` not installing `talker.py`).

racko, 2017-10-25:
The following line fails when you build roslaunch using only the install space of rospy instead of the devel space:
https://github.com/ros/ros_comm/blob/8966ac6c954edd078825ff6f44a1bd69847d833b/tools/roslaunch/test/unit/test_nodeprocess.py#L128

I thought it was a good idea to run the tests for freshly built ros packages before packaging for Arch Linux. But building https://github.com/ros-gbp/ros_comm-release/archive/release/lunar/roslaunch/1.13.2-0.tar.gz while sourcing /opt/ros/lunar/setup.bash leads to the error above, because rospy does not install the test nodes (i.e. talker.py).

But if @mikepurvis's statement is generally accepted, then I guess I will just not run those tests ;)

mikepurvis, 2017-10-25:
You can still build the install space and run your tests against the devel space, but that's going to be trickier if you're building package by package rather than repo-at-a-time, since as you've identified, the roslaunch tests don't only depend on roslaunch itself being in the devel space, but _also rospy being in the devel space_.

dirk-thomas, 2017-10-25:
Since it is not desired to install these test nodes in the `rclpy` package I think the right way to fix this is to update `roslaunch` to either change to different nodes which are available (not sure which ones that could be) or copy the talker and listener scripts into the `roslaunch` package in order to be available for testing.

racko, 2017-11-03:
Since few ros packages provide tests (most are in separate *_test packages) I don't intend to run tests during packaging anymore. Since the issue I describe above isn't much of a bug, I close this issue.

closed 2017-11-03 by racko