ros/ros_comm #1203: ros::NodeHandle::subscribe interface bug?

AdamDorwart, 2017-10-25:
Maybe I'm just not seeing something but [this method](http://docs.ros.org/api/roscpp/html/classros_1_1NodeHandle.html#a302620aff50f66c4b73fc613a55c27aa) of NodeHandle looks like it has an extraneous template argument.

Taking a look at the method (body)[https://github.com/ros/ros_comm/blob/lunar-devel/clients/roscpp/include/ros/node_handle.h#L798-L807]
```C++
template<class M, class C>
Subscriber subscribe(const std::string& topic, uint32_t queue_size, const boost::function<void (C)>& callback, const VoidConstPtr& tracked_object = VoidConstPtr(), const TransportHints& transport_hints = TransportHints())
{
  SubscribeOptions ops;
  ops.template initByFullCallbackType<C>(topic, queue_size, callback);
  ops.tracked_object = tracked_object;
  ops.transport_hints = transport_hints;
  return subscribe(ops);
}
```
There doesn't seem to be any usage of M. With gcc 5.4.0 I get
> error: no matching function for call to ‘ros::NodeHandle::subscribe(...)
> /opt/ros/kinetic/include/ros/node_handle.h:798:14: note: candidate: template<class M, class C> ros::Subscriber ros::NodeHandle::subscribe(const string&, uint32_t, const boost::function<void(C)>&, const VoidConstPtr&, const ros::TransportHints&)
>note: couldn't deduce template parameter ‘M’

If I provide ANY parameter type for M it will compile.
`nh.subscribe<int>(topic, queue_size, func);`

racko, 2017-11-03:
I tried to do the obvious and remove the template parameter `M`. The following test fails to compile:
https://github.com/ros/ros_comm/blob/acb152c34d1210bb7ca6ed230e0f8b9389eb453d/test/test_roscpp/test/src/subscription_callback_types.cpp#L171
Removing the obsolete first template argument (std_msgs::String) leads to more compiler errors.
The error messages are hard to read because they only say that some boost::function constructor call is "forming a pointer to a reference". But my IDE helped explain the issue: The subscribe overload being called is not the one on line 798 but the one on 750.

I'm not quite sure why that one is considered a better match than the other one. I would have expected at least an "ambiguous function call" error.

But ignoring that question, it turns out that the "unnecessary" template argument currently serves as a disambiguation hint.

dirk-thomas, 2018-02-03:
Based on you second command I assume there is nothing to be done at the moment hence I will close this ticket for now. Please feel free to continue commenting and and if necessary the issue can be reopened.

closed 2018-02-03 by dirk-thomas