ros/ros_comm #1207: non-portable code in rosmaster.util

Yay295, 2017-10-28:
The last change to `util.py` ( #1104 ) added a function `close_half_closed_sockets()` to fix an issue where some sockets weren't being closed, causing ROS to eventually freeze up. Unfortunately, the call `getsockopt(socket.SOL_TCP, socket.TCP_INFO)` is not portable ([man page](http://man7.org/linux/man-pages/man7/tcp.7.html) "**TCP_INFO** Used to collect information about this socket. The kernel returns a struct tcp_info as defined in the file /usr/include/linux/tcp.h. *This option should not be used in code intended to be portable.*"), and it fails under WSL. There is a [WSL Issue](https://github.com/Microsoft/WSL/issues/1982) tracking this, and it appears they do intend to implement this functionality, but I thought it would be a good idea to bring it up here as well.

dirk-thomas, 2017-10-30:
Does this also affect other platforms like native Windows?

@otamachan @furushchev Does either of you see a way to avoid this socket option?

otamachan, 2017-10-30:
It seems that WIndows does not support `TCP_INFO`

```
c:\Python27>python
Python 2.7.14 (v2.7.14:84471935ed, Sep 16 2017, 20:19:30) [MSC v.1500 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import socket
>>> socket.TCP_INFO
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'TCP_INFO'
```

I'm not familiar with Windows sockets, so currently I have no idea how to avoid `TCP_INFO`.

furushchev, 2017-10-31:
It looks that windows has an api for getting TCP socket states in C++ level but I'm not sure if this functionality is accessible from `python` on windows.

https://msdn.microsoft.com/en-us/library/bb485738(VS.85).aspx

For now, I can only suggest not to use this function on the system that does not support `TCP_INFO` until any solution is found since I have no experience with windows.

smnogar, 2017-12-21:
I am getting issues with this in macOS High Sierra as well. Errors appear in master.log that look like: 
```
[rosmaster.threadpool][ERROR] 2017-12-21 15:49:02,582: Traceback (most recent call last):
  File "/opt/ros/lunar/lib/python2.7/site-packages/rosmaster/threadpool.py", line 218, in run
    result = cmd(*args)
  File "/opt/ros/lunar/lib/python2.7/site-packages/rosmaster/master_api.py", line 210, in publisher_update_task
    ret = xmlrpcapi(api).publisherUpdate('/master', topic, pub_uris)
  File "/opt/ros/lunar/lib/python2.7/site-packages/rosmaster/util.py", line 68, in xmlrpcapi
    close_half_closed_sockets()
  File "/opt/ros/lunar/lib/python2.7/site-packages/rosmaster/util.py", line 77, in close_half_closed_sockets
    state = transport._connection[1].sock.getsockopt(socket.SOL_TCP, socket.TCP_INFO)
AttributeError: 'module' object has no attribute 'TCP_INFO'
```

Reverting `util.py` to [this](https://github.com/ros/ros_comm/commit/2d987bcbb46768168dccdd58f35bdee8b705d52a#diff-1b19eb57ea1f1e3a593df181f583baeb) makes everything work fine again.

Yay295, 2017-12-22:
The fix for this has been merged and released on Ubuntu at least. Have you updated ROS recently @smnogar?

smnogar, 2017-12-22:
@yay295, yeah, I rebuilt both lunar and kinetic today with all the latest dependencies available on brew. I haven't seen it mentioned anywhere else so maybe its some soft of conflict brought about by recent python updates or something. Maybe its just me.

otamachan, 2017-12-22:
Hi, according to the error messages,  I think #1211 will help. I've rebased #1211 just now. Could you test it ?

smnogar, 2017-12-23:
@otamachan, it worked for me. No problems now.

dirk-thomas, 2018-02-03:
Addressed in #1211.

closed 2018-02-03 by dirk-thomas