ros/ros_comm #1208: Rospy for Python 3

rivermont, 2017-10-30:
It seems the only way to use Python 3.x with rospy is to rebuild all of the ROS packages from source, forcing Python 3. However this seems like it would cause a lot of compatibility issues. Is there any other way to create nodes in Python 3 for ROS?

dirk-thomas, 2017-10-30:
Using a consistent version of Python throughout all packages is certainly the recommended way.

If you know that all recursive Python code of your node works with Python 2 as well as 3 you can just use a Python 3 shebang line for your node and as long as all dependencies are being found your node should work fine. That "works" because ROS add the Python code of all ROS packages to the `PYTHONPATH`. For some system dependencies (like `catkin-pkg-modules`, `python-yaml`) you will have to also install the Python 3 equivalent.

In general since this is more a question rather than an action item (like a bug, feature request, etc.) I will close this since there is nothing to do about it in this repo. In the future we kindly ask to raise questions like this on [answers.ros.org](http://answers.ros.org/) instead.

closed 2017-10-30 by dirk-thomas

dirk-thomas, 2019-02-04:
> `python3-catkin-pkg-modules` cannot be installed because the Python2 version is installed.

That statement is incorrect. `python-catkin-pkg-modules` and `python3-catkin-pkg-modules` can absolutely be installed side-by-side. These packages only contain the Python modules of `catkin_pkg` but not the command line tools.

The packages `python-catkin-pkg` and `python3-catkin-pkg` can't be installed side-by-side since they provide the same command line tools - just with a different shebang line.

Neuan, 2019-05-02:
Any suggestion on how to install rospy inside a Python3 virtual environment?
Thanks.