ros/ros_comm #1209: ROSBagException should call the base constructor

lukehansen, 2017-10-30:
In python 2.7 there is a subtle bug with unpickling custom exceptions; this was fixed in 3.3, see https://bugs.python.org/issue1692335.  As a workaround for 2.7 we can re-define these exceptions as below.

```java
class ROSBagException(Exception):
    """
    Base class for exceptions in rosbag.
    """
    pass
 
class ROSBagFormatException(ROSBagException):
    """
    Exceptions for errors relating to the bag file format.
    """
    pass

class ROSBagUnindexedException(ROSBagException):
    """
    Exception for unindexed bags.
    """
    def __init__(self, *args):
        ROSBagException.__init__(self, 'Unindexed bag')
```

For context, I came across this issue when a `ROSBagUnindexedException` was raised in a `multiprocessing.Pool.map` thread.  It throws the `__init__` exception and hangs, as described here: https://lists.gt.net/python/bugs/1025933

dirk-thomas, 2017-10-30:
Can you please provide a pull request with the proposed patch against the current default branch (`lunar-devel`). Thanks.

lukehansen, 2017-10-30:
Sure, though it's not really a ros_comm bug and has been patched in python 3.3.  PR is here: https://github.com/ros/ros_comm/pull/1210

PR #1652 (2019-04-01): Pickleable rosbag exceptions (#1210 revisited).
This fixes  #1209 in a API preserving way. It replaces #1210. Also tests for the pickleability have been added.
closed 2019-04-01 by dirk-thomas