ros/ros_comm #1222: rospy.log*_throttle (and _once) do not accept (kw)args

gavanderhoorn, 2017-11-08:
It would appear the `logX_throttle(..)` and `logX_once(..)` methods introduced in #812 and #1041 do not accept args or keyword args like the regular `logX(..)` methods do.

This means that instead of `rospy.loginfo("fmt", arg0, arg1, argN)` they have to be passed a bare string (ie: `rospy.loginfo_throttle(1.0, "some string")`).

This can be worked around by using `str.format(..)` itself, but symmetry with the base `logX(..)` methods is lost this way.

Was this intentional, or an oversight?

dirk-thomas, 2017-11-08:
@wkentaro Maybe you can answer the question and/or provide a follow up PR to align the signatures.

dirk-thomas, 2020-02-10:
Addressed by #1290.

closed 2020-02-10 by dirk-thomas