ros/ros_comm #1229: Message deserialization regression

mikepurvis, 2017-11-10:
Still trying to pinpoint exactly where this is coming from, but just wanting to fire an early flare—we're seeing a ton of spam like this in our syslog:

```
$ sudo journalctl -n 10 -u robot
-- Logs begin at Fri 2017-11-10 14:49:40 UTC, end at Fri 2017-11-10 15:31:03 UTC. --
Nov 10 15:30:48 robot-44 robot-start[1978]: [ERROR] [/rzr_node] [1510327848.077547916]: Exception thrown when deserializing message of length [8] from [/rzr_node]: Buffer Overrun
Nov 10 15:30:48 robot-44 robot-start[1978]: [ERROR] [/rzr_node] [1510327848.099453371]: Exception thrown when deserializing message of length [8] from [/rzr_node]: Buffer Overrun
Nov 10 15:30:48 robot-44 robot-start[1978]: [ERROR] [/rzr_node] [1510327848.099659662]: Exception thrown when deserializing message of length [8] from [/rzr_node]: Buffer Overrun
```

This just started in the last couple days, and we're building from `lunar-edge` as of da17ee01763011c3d4367924a8dfa9dc65d0ca83.

Setting a breakpoint and [jumping up the stack to this point](https://github.com/ros/ros_comm/blob/lunar-devel/clients/roscpp/src/libros/message_deserializer.cpp#L78), it's possible to get a little more context:

```
(gdb) p params
$4 = {buffer = 0x1562264 "", length = 4, connection_header = {px = 0x155a9e0, pn = {pi_ = 0x155d4e0}}}
(gdb) p serialized_message_
$5 = {buf = {px = 0x1562260 "\004", pn = {pi_ = 0x1563e90}}, num_bytes = 8, message_start = 0x1562264 "", message = {px = 0x0, pn = {pi_ = 0x0}}, type_info = 0x0}
(gdb) p *connection_header_.px
$23 = std::map with 6 elements = {["callerid"] = "/rzr_node", ["latching"] = "0", ["md5sum"] = "fdb28210bfa9d7c91146260178d9a584", ["message_definition"] = "float64 data\n", 
  ["topic"] = "/[redacted]", ["type"] = "std_msgs/Float64"}
```

So the topic in question is a `Float64`, and based on the topic name, we know the originating end is also roscpp.

mikepurvis, 2017-11-10:
Okay, using LD_PRELOAD, I tried running our robot with several previous versions of `libroscpp.so`, and I still see the problem on 1.13.2 (2cb31a02966628e0de0a61873cf8b0bb589b164d), which was from August. So the problem is either:

- Something in a header, so I'd need to recompile more of our system to get the fix associated with rolling back.
- Something else going on on our end.

mikepurvis, 2017-11-10:
Disregard. This has been traced to some user code on our side.

closed 2017-11-10 by mikepurvis

dirk-thomas, 2017-11-10:
Thank you for investigating!

mikepurvis, 2017-11-10:
The root cause turned out to be that a well-meaning individual had turned logic like this:

    std_msgs::Float64 out_msg;
    out_msg.data = msg.measured_velocity;  // a field of type float32
    topic_.publish(out_msg);

Into this:

    topic_.publish(msg.measured_velocity);

It seems like the compiler just rolled with the traits supplied in [std_msgs/builtin_float.h](https://github.com/ros/std_msgs/blob/groovy-devel/include/std_msgs/builtin_float.h#L34), and the message ended up published despite being the wrong size and the wrong type.

This seems like an easy mistake to make, and it wasn't at all obvious what the issue was. Is there some way the builtin_xx message headers can be made less dangerous?