ros/ros_comm #1230: Buffer size check will always succeed

sjenkins-cp, 2017-11-13:
https://github.com/ros/ros_comm/blob/f3c4ae8c80b557dd44fac3ed8797cd73d771d0b9/clients/roscpp/src/libros/connection.cpp#L121

The condition on line 122 will always be true because to_read is an unsigned integer. This may lead to some nasty buffer allocation issues if it ever should have been negative (which is actually an issue I am trying to track down when I came across this).

dirk-thomas, 2017-11-13:
Can you please explain why the condition `to_read > 0` would always be `true`? In the case `read_size_` and `read_filled_` have the same value `ro_read` is zero which makes the condition `false`.

sjenkins-cp, 2017-11-13:
Actually, == 0 is legit so this may be as intended. Going to close until I have followed this through further.

sjenkins-cp, 2017-11-13:
Yep apologies for that - no one spotted that went we discussed.

closed 2017-11-13 by sjenkins-cp