ros/ros_comm #1231: Change Color of Rosconsole

nathangeorge1, 2017-11-13:
Hello,

Could a feature be implemented to change the color of the various logger levels, or even a mode like ROS_INFO_COLOR where one could specify the color?

Thanks,
Nathan

dirk-thomas, 2017-11-13:
The logging output is already being colorized. Please try running either of the following nodes without starting a ROS core: `rosrun roscpp_tutorials talker` / `rosrun rospy_tutorials talker`.

nathangeorge1, 2017-11-13:
Hello @dirk-thomas,

I have worked with the red, orange, green, and white colors of rosconsole. My question is if a feature could be added so that the logger levels can be changed programmatically?

Thanks,
Nathan

dirk-thomas, 2017-11-13:
Oh, I understand now. Currently the color values are hardcoded (e.g. in [C++](https://github.com/ros/ros_comm/blob/93287768d44d741101c69594f52498c3e26538b3/tools/rosconsole/src/rosconsole/rosconsole.cpp#L103-L106)). Please consider creating a PR for this feature idea. It is unlikely that the maintainers will be able to work on this.

nathangeorge1, 2017-11-13:
ok; do you think if I create a PR, it is likely to be accepted? Are there any requirements I need to meet?

dirk-thomas, 2017-11-13:
Yes, pull requests usually get accepted. To increase the chances and reduce the effort on iterating on the patch I would recommend to:

* target the latest development branch (currently `lunar-devel`)
* keep the changes small, avoid any unrelated changes
* maintain backward compatibility
* add unit tests covering the new feature
* after the PR has been merged add information about the new feature to the wiki page(s)

nathangeorge1, 2017-11-13:
ok will do; a few questions:
1. I work with kinetic; if I work on lunar, then will kinetic get the changes?
2. Where do I add unit tests?

dirk-thomas, 2017-11-13:
> 1. I work with kinetic; if I work on lunar, then will kinetic get the changes?

After changes have been merged and released into the latest distro (atm `lunar`) they will be considered for backporting. New features are not automatically being backported. There needs to be a strong incentive for providing the new feature in already existing distros to do so since there is always a chance for regressions. The cleaner the patch is and the easier it is to "verify" that it should be fine the higher are chances to be backported into Kinetic in this case. Backporting such a feature to Indigo is very unlikely though.

> 2. Where do I add unit tests?

Most likely in a `test` folder within the package you are adding the feature to.

nathangeorge1, 2017-11-13:
great; thank you for your guidance. I don't know when I'll create this PR, but I'll surely do it sometime soon or in the coming months

nathangeorge1, 2018-01-27:
So it seems that while color is important, I don't think I need it now; and because time is short, I'll probably revisit this at a later time, if that works.

nathangeorge1, 2018-02-23:
I recently found out that an easier way to handle this is to make a wrapper around the ROS_INFO and other print out functions that support color. I think I'll do this as I'm not familiar with the ROS codebase yet. Thanks for your help.

closed 2018-02-23 by nathangeorge1

oscar-lima, 2023-12-18:
the wrapper idea is a good one, here is a snippet that works well:

    def loginfo_green(message):
        '''
        This function acts as a wrapper around rospy.loginfo to print messages in green color.
        '''
        green_start = '\033[92m'
        color_reset = '\033[0m'
        rospy.loginfo(green_start + str(message) + color_reset)