ros/ros_comm #1240: Rosbag improvement proposal: a better alternative to --clock and use_sim_time=true

facontidavide, 2017-11-19:
Hi,

as it happens to hundred of users, when I play rosbags it is somehow complicated to make things work correctly using in terms of timestamps, being TF a notable example.

The official workaround is to use __"rosbag play --clock"__ alongside to __use_sim_time=true__.

I believe I can contribute with a better solution. Thanks to [ros_type_introspection ](http://wiki.ros.org/ros_type_introspection) I am able to know at run-time if any message has a __Header__ field and I can inject into the message a new timestamp before publishing it as __topic_tools::ShapeShifter__.

This new timestamp can be consistent with ros::Time::now() in terms of delay from the beginning of the rosbag.

In other words, message will appear to be generated in real-time, rather than being recorded.

Do you think that this functionality is desirable? 

Davide

dirk-thomas, 2018-02-03:
Adding an option for this feature would certainly be helpful in some cases. I have only briefly looked at `ros_type_introspection` and it depends on `rosbag` itself so I am not sure how `rosbag` can use it (basically it needs to invert the dependency direction). Please feel free to provide a pull request for this and I am happy to review / merge it.

NikolausDemmel, 2018-03-05:
@facontidavide, this sounds like a great addition. In the past I had multiple times wished I had something like that. Two comments:

* Changing header.stamp might work for many default cases, but some (custom) messages might contain additional timestamps. You could include support for some standard cases, like e.g. `/tf` (where the messages don't have headers, but contain a list of Transform messages with headers to update), but in general it will be impossible to cover everything. Even without it would be useful of course, but maybe you can think about possible solutions (maybe some sort of extension mechanism to define rules for updating `Time` fields in custom messages).
* For many use cases where the timestamps matter, it is actually fairly important that the relative time between the messages doesn't change. If you just put `ros::Time::now()` that will not be given. Solving this is not so trivial. You can compute a constant offest at the beginning and then subtract that before publishing (instead of replacing with `now()`), but what do you do if publishing cannot be done fast enough? Then your stamps and the system time will again fall out-of-sync. Again, it would probably be useful even with this limitation.

dmcconachie, 2018-04-04:
In addition this would be particularly useful when using multiple bag files, recorded at different times. If you want them to play with timestamps in a similar range, then right now a custom republisher that replaces timestamps is the only way to go that I am aware of.

NikolausDemmel, 2018-04-05:
>  right now a custom republisher that replaces timestamps is the only way to go that I am aware of.

Just as a side note: Not sure what you mean by a `republisher` (multiple `rosbag play` and "republisher" nodes to update timestamps?), but my approach would be to have small python script that combines the bag files into a single one with adjusted timestamps, and then play that one bag file back.

facontidavide, 2018-04-06:
I am happy to see some interest. I will prepare a PR and submit it for review very soon.

Cheers

facontidavide, 2018-04-12:
Hi,

good news and bad news.

I achieved a working prototype, so __technically__ it is very possible.
Bad news are that my changes are probably impossible to merge because it would make rosbag dependent on:

- ros_type_introspection: I removed the circular dependency, but it requires C++11
- abseil_cpp: very nice but fat-ass dependency.

I guess that the best would be to create a separate project and share it with the community, rather than trying to merge these changes with __ros_comm__.

Feel free to close this issue.

NikolausDemmel, 2018-04-12:
Cool, sounds like these dependencies might be a problem for merging to ros_comm.

Providing a separate package could a good alternative. I'm not sure if there is an extension mechanism for the rosbag API (if there is not, it could be added) that would allow your package to even register a new verb like `rosbag playlive ...` if your package is installed.

facontidavide, 2018-04-12:
sounds a great way to move this forward... and I like the "playlive" command ;)

I will keep the community updated on Discourse. Cheers

closed 2018-04-12 by facontidavide

NikolausDemmel, 2018-04-12:
Cool. Thanks for your effort! It would be great if you could post a link here at some point.

lucasw, 2022-02-12:
I just read https://discourse.ros.org/t/timestamps-and-rosbags-discussing-an-alternative-to-clock-and-use-sim-time/3238/9 and then looked around and got to here, then saw up in that same thread there's a python script to update timestamps https://github.com/DLu/wu_ros_tools/blob/noetic/rosbaglive/scripts/play.py (also http://wiki.ros.org/rosbaglive)

It only descends into the top level `header` and does do any looking for deeper down header.stamps to update, except in the case where there is a list of `transforms` it'll get those.

It also doesn't factor in offsets between the header stamp and rosbag time-of-reception stamp, but that wouldn't be hard to add.