ros/ros_comm #1245: Rosbag python3 troubles

akru, 2017-11-22:
With `rosbag` on Python3 I have an issue like this:

```
  File "/home/akru/httpbag/venv/lib/python3.6/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/akru/httpbag/venv/lib/python3.6/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/akru/httpbag/venv/lib/python3.6/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/akru/httpbag/venv/lib/python3.6/site-packages/flask/_compat.py", line 33, in reraise
    raise value
  File "/home/akru/httpbag/venv/lib/python3.6/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/akru/httpbag/venv/lib/python3.6/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "./httpbag.py", line 52, in main
    content = content + bag_to_str(p)
  File "./httpbag.py", line 19, in bag_to_str
    for topic, msg, t in bag.read_messages():
  File "/nix/store/skbmpcggfql9360xwd6pvag99lrwh1gf-rosbag/lib/python3.6/site-packages/rosbag/bag.py", line 2337, in read_messages
    yield self.seek_and_read_message_data_record((entry.chunk_pos, entry.offset), raw)
  File "/nix/store/skbmpcggfql9360xwd6pvag99lrwh1gf-rosbag/lib/python3.6/site-packages/rosbag/bag.py", line 2446, in seek_and_read_message_data_record
    raise ROSBagException('unsupported compression type: %s' % chunk_header.compression)
rosbag.bag.ROSBagException: unsupported compression type: b'none'
```

As I can see exception rises cause `b'none' != 'none'`.

dirk-thomas, 2017-12-18:
Please provide more information (e.g. which version you are using) as well as a reproducible example.

akru, 2017-12-18:
As I can see, it fixed in latest release, thank you!

closed 2017-12-18 by akru