ros/ros_comm #1248: Source builds on Kinetic fail with XmlRpcValue.h inclusion problem

gavanderhoorn, 2017-11-27:
As reported in https://github.com/ros/ros_comm/pull/962#issuecomment-347181447:

While trying to build Kinetic `desktop-full` from sources on Ubuntu Xenial (amd64) I'm running into a header inclusion problem with `XmlRpcValue.h` in `diagnostic_aggregator`:

```
In file included from /catkin_ws/src/diagnostics/diagnostic_aggregator/src/analyzer_group.cpp:37:0:
/catkin_ws/src/diagnostics/diagnostic_aggregator/include/diagnostic_aggregator/analyzer_group.h:51:25: fatal error: XmlRpcValue.h: No such file or directory
compilation terminated.
make[2]: *** [CMakeFiles/diagnostic_aggregator.dir/src/analyzer_group.cpp.o] Error 1
```

This seems to be the same issue that is reported in https://github.com/ros/geometry/issues/144#issuecomment-335630911. It should probably be fixed in `ros/diagnostics`, but seeing as the change introduced in #962 was supposed to be bw-compatible, I figured I report it here as well.

Adding `include/xmlrpcpp` to the `catkin_package(..)` `INCLUDE_DIRS` in `ros_com/xmlrpcpp/CMakeLists.txt:12` does seem to make the build continue, but that should not be necessary.

gavanderhoorn, 2017-11-27:
After adding an `message (WARNING ..)` to `xmlrpcpp-extras.cmake.in` it would seem it is adding a non-existing path (at least on my system) to the include dirs.

With a workspace at `/catkin_ws`, it adds: `/catkin_ws/devel/share/xmlrpcpp/cmake/../../../include/xmlrpcpp`, but that doesn't exist at that point.

I'm not sure whether this could be a timing / ordering issue with `catkin_tools` btw. Having `catkin_tools` build just `xmlrpcpp` doesn't result in the `include` dir appearing.

gavanderhoorn, 2017-11-27:
@gavanderhoorn wrote:
> I'm not sure whether this could be a timing / ordering issue with `catkin_tools` btw. Having `catkin_tools` build just `xmlrpcpp` doesn't result in the `include` dir appearing.

Nope. Building with `catkin_make_isolated` results in the same inclusion error. `/catkin_ws/devel_isolated/xmlrpcpp/share/xmlrpcpp/cmake/../../../include/xmlrpcpp` doesn't exist at that point.

gavanderhoorn, 2017-11-27:
@jspricke: seeing as you added the `CFG_EXTRAS` file: how did you come up with the expression for the path?

jspricke, 2017-11-27:
Good question. Honesty I don't remember anymore.

kentsommer, 2017-12-12:
Just chiming in that I saw this same error building `diagnostic_aggregator` when building ROS Kinetic Desktop-Full from source on a fresh Ubuntu 16.04.3 install. I am positive that the required dependencies were installed correctly (namely `libxmlrpcpp-dev`).

dirk-thomas, 2017-12-13:
The CMake extra file provided in #962 does only work correctly when installing the `xmlrpcpp` package but not when it is only in a devel space. Please see #1261 for the necessary fix. It would be great if one of you could try the patch and confirm that it fixes the problem for you too.

closed 2017-12-13 by dirk-thomas

kentsommer, 2017-12-14:
@dirk-thomas 

Just commenting to say this does indeed fix the issue. :+1:

dirk-thomas, 2017-12-14:
> Just commenting to say this does indeed fix the issue

@kentsommer Thank you for checking and reporting back.

kenplusplus, 2017-12-17:
@dirk-thomas the patch #1261 really fixed the issue on Luna branch, but kinetic 1.12.12-0 is still broken, which will impact the "desktop rosinstall" generated by rosinstall_generator. Would you have plan to back port it on Kinetic release?

kentsommer, 2017-12-17:
@kenplusplus 

I just submitted a PR for the backport (#1269).

bluewish, 2017-12-17:
@kentsommer you are so quick! 👍

PR #1261 (2017-12-13): make xmlrpcpp specific include directory work in devel space
Fixed #1248. Needs to be backported to Kinetic.