ros/ros_comm #1251: Consolidate ROS_HOSTNAME and ROS_IP

130s, 2017-12-09:
There's a [q248415](https://answers.ros.org/question/248415/ros_ip-and-ros_hostname/) opened by @otamachan, where he also suggests to only use `ROS_HOSTNAME`.

The example section in the [NetworkSetup wiki](http://wiki.ros.org/ROS/NetworkSetup#Setting_a_name_explicitly) explains when the both variables are useful, but the usecases don't sound convincing that we have to have 2 different variables. Are these actually usecases where separate variables are needed?

If not, I'm basically in favor of @otamachan's idea. In that case we should deprecate `ROS_IP`, print deprecation notice, update documents as well IMO.

gavanderhoorn, 2017-12-09:
I think the use-cases are pretty clear: have a working DNS, but your system is reporting a non-resolvable name? Use `ROS_HOSTNAME` to make sure only a resolvable hostname is reported to all parts of the infrastructure that are impacted by that. If you have a DHCP server on top of that, your ip might change, but that doesn't matter, as DNS should still resolve it properly.

Now for `ROS_IP`: don't have a working DNS, no DHCP (or static assignements) and multi-homed system(s)? Use `ROS_IP` to again make sure the correct info (in this case the ip) is communicated to all participants in your node graph. Multi-homed systems are actually quite common these days (cellular tethering, machines with both wired and wireless, etc), and networks without DNS (or a DNS that doesn't resolve for all hosts on a network, such as a home router) as well. `ROS_IP` covers these cases.

As for why we'd keep both: if you'd use `ROS_IP` (or set `ROS_HOSTNAME` to an ip) and DHCP assigns you a different ip, you'd have to make sure you update your env vars or else things stop working. This is actually quite a common problem. See [#q201225](https://answers.ros.org/question/201225) on ROS Answers for an example Q&A.

Personally I like the fact that there are two variables: semantics for both are different, and it makes it explicit that the two should be used to configure different things. DNS libraries may return immediately if they are requested to 'resolve' an IP address, but that doesn't mean `ip address == hostname`.

jspricke, 2017-12-09:
I vote for deprecating `ROS_HOSTNAME`. I don't agree that both variables have different semantics. Hostnames and IP addresses should be interchangeable. If they are not, your DNS is broken and IP is your best bet anyhow ;). So instead of having one taking precedence over the other (something I've seen people struggling with quite often), simply document that you could put in both (and IP usually works best).

gavanderhoorn, 2017-12-09:
@jspricke wrote:
> Hostnames and IP addresses should be interchangeable.

except they're not.

Hostnames add a layer of indirection that can be exploited to get more flexibility. IP addresses are final: there is no indirection possible, and an incorrect/stale value will mean things won't work.

jspricke, 2017-12-09:
* G.A. vd. Hoorn <notifications@github.com> [2017-12-09 11:38]:
>Hostnames add a layer of indirection that can be exploited to get more flexibility. IP addresses are final: there is no indirection possible, and an incorrect/stale value will mean things won't work.

Can you name an example where this makes a different to ROS and where 
it's not solvable with on variable? Thanks!

otamachan, 2017-12-11:
Hi @130s,
thanks for opening a ticket.

IMO, it is not a big problem that both variables exist. I think the problem is that there is no warning even when both variables are set, which makes ROS network trouble shooting a bit harder for beginners because they sometimes set both variables accidentally.
So I think adding a warning rather than deprecating ROS_IP might be enough.
Thanks.

dirk-thomas, 2018-02-03:
I don't think that the existence of the two variables is an issue. They are both documented and their names is pretty clear in terms of what they should contain.

If someone would like to work towards consolidating these two variables into a single one please feel free to contribute for that effort. Just deprecating one of them will certainly not be enough though. If you want to move forward on this please address all relevant aspects ranging from a PR to deprecate one variable in an upcoming ROS distro, a PR to remove one variable in a future release, updates to various wiki pages mentioned the current variables.

dirk-thomas, 2020-08-03:
With Noetic being released and likely being the last ROS 1 distro and this ticket proposing to change behavior which shouldn't target an already released distro I will go ahead and close it.

closed 2020-08-03 by dirk-thomas