ros/ros_comm #1262: ROS_IPV6 is checked too late in roscpp

zielmicha, 2017-12-13:
ROS_IPV6 environment variable is checked late in initialization process (in `ros::start`):
```cpp
env_ipv6 = getenv("ROS_IPV6"); //...
bool use_ipv6 = (env_ipv6 && strcmp(env_ipv6,"on") == 0); //...
XmlRpc::XmlRpcSocket::s_use_ipv6_ = use_ipv6;
```

However, first connection to master can be initialized as early as in `ros::init` - failing the whole process and preventing nodes from starting. 

If I change default value of `XmlRpc::XmlRpcSocket::s_use_ipv6_` to `true` manually, these nodes work in IPv6 environment. The correct solution would be to move ROS_IPV6 check to the beginning of `ros::init`.

dirk-thomas, 2017-12-13:
Can you please provide a reproducible example. It would be great if you could  consider to create a pull request for addressing this too.

zielmicha, 2017-12-15:
1. Run roscore on machine A with `ROS_IPV6=on`.
2. Run rviz on machine B (with `ROS_MASTER_URI` set to machine A and `ROS_IPV6=on`). Observe that rviz can't connect to master.

Here is a patch that fixes this problem - I'm not sure which branch I should submit PR to:

```patch
diff --git a/clients/roscpp/src/libros/init.cpp b/clients/roscpp/src/libros/init.cpp
index 91e0a58..7a38aaa 100644
--- a/clients/roscpp/src/libros/init.cpp
+++ b/clients/roscpp/src/libros/init.cpp
@@ -316,17 +316,6 @@ void start()
     }
   }
 
-  char* env_ipv6 = NULL;
-#ifdef _MSC_VER
-  _dupenv_s(&env_ipv6, NULL, "ROS_IPV6");
-#else
-  env_ipv6 = getenv("ROS_IPV6");
-#endif
-
-  bool use_ipv6 = (env_ipv6 && strcmp(env_ipv6,"on") == 0);
-  TransportTCP::s_use_ipv6_ = use_ipv6;
-  XmlRpc::XmlRpcSocket::s_use_ipv6_ = use_ipv6;
-
 #ifdef _MSC_VER
   if (env_ipv6)
   {
@@ -430,6 +419,19 @@ end:
   }
 }
 
+void check_ipv6_environment() {
+  char* env_ipv6 = NULL;
+#ifdef _MSC_VER
+  _dupenv_s(&env_ipv6, NULL, "ROS_IPV6");
+#else
+  env_ipv6 = getenv("ROS_IPV6");
+#endif
+
+  bool use_ipv6 = (env_ipv6 && strcmp(env_ipv6,"on") == 0);
+  TransportTCP::s_use_ipv6_ = use_ipv6;
+  XmlRpc::XmlRpcSocket::s_use_ipv6_ = use_ipv6;
+}
+
 void init(const M_string& remappings, const std::string& name, uint32_t options)
 {
   if (!g_atexit_registered)
@@ -453,6 +455,7 @@ void init(const M_string& remappings, const std::string& name, uint32_t options)
 #ifndef WIN32
     signal(SIGPIPE, SIG_IGN);
 #endif
+    check_ipv6_environment();
     network::init(remappings);
     master::init(remappings);
     // names:: namespace is initialized by this_node
```

dirk-thomas, 2017-12-15:
> I'm not sure which branch I should submit PR to

Please create a pull request against the default branch (currently `lunar-devel`).

commit 970372df (2018-01-16): Fixes #1262 - bad IPv6 initialization order. (#1268)

ROS_IPV6 environment variable was checked after first connection could be established.
This commit moves this check into early initialization.
commit b02278fb (2018-02-09): Fixes #1262 - bad IPv6 initialization order. (#1268)

ROS_IPV6 environment variable was checked after first connection could be established.
This commit moves this check into early initialization.
closed 2018-01-16 by dirk-thomas