ros/ros_comm #1263: [roslaunch_add_file_check] machine tag in launch file

bmagyar, 2017-12-13:
A `<machine>` tag conceptually invalidates all checks done on a launch file.
This makes it impossible to use roslaunch validation on files deployed on machines of a different platform (RPI packages in my case). 

I started digging into the code to see if there's a parameter for this somewhere but couldn't find any. Do you agree with skipping (perhaps with a warning) parsing files with the `<machine>`  tag? Should it be default or optional?

mikepurvis, 2017-12-13:
AFAIK the motivating use case for `<machine>` was to launch stuff on PR2's dual servers, one of which mounted the other's disk as an NFS root or something. I don't really know the details, but `<machine>` has always lived in a nebulous world where it doesn't actually require that kind of underlying shared-disk setup, but if you don't happen to have it, a whole bunch of stuff about the behaviour of it makes a lot less sense.

I don't have an opinion on what to do about machine tags in the checker. We don't use them, so it doesn't impact me one way or another.

bmagyar, 2017-12-18:
That's a cool backstory, thanks for sharing! :) 

Perhaps @dirk-thomas has the best overview of `roslaunch` to answer this?

tfoote, 2017-12-19:
The machine tag is useful for any distributed environment. You don't necessarily need full NFS synchronization but you do need to know what the target development environment will be like, to tell roslaunch what `env-loader` to use. It does mean that you cannot check that the other machine is accessible and setup correctly. However, it could be asserted that the validation tests would expect the same version of packages on the other end and thus you could check that the launch files will validate if the other end has the same version. If there's mismatched versions between machines that's on the system administrator.

bmagyar, 2017-12-19:
That's exactly my question. What happens when different machines run
different architectures?

My CI runs on an x64 server while there are some launch files using the
machine tag that get started on a raspberry pi.  Some packages referred by
these launch files use nodes not available on x64 hence the checker fails
on them.

dirk-thomas, 2017-12-19:
> That's exactly my question. What happens when different machines run
different architectures?
> My CI runs on an x64 server while there are some launch files using the
machine tag that get started on a raspberry pi.  Some packages referred by
these launch files use nodes not available on x64 hence the checker fails
on them.

If the system you are running the tests on is not able to run the launch files started using the `machine` tags you might not be able to use the `roslaunch_add_file_check` functionality.

mikepurvis, 2017-12-20:
Depending on the functionality you have on the Pi, it may make sense to just launch that in its own local systemd job, or put a trivial supervision script around it that launches it when the remote master is available and shuts it down when it disappears.

In my limited exposure to multi-computer robots, that ended up being the preferred approach vs. trying to get the machines closely enough aligned to use `<machine>`.

bmagyar, 2017-12-20:
Thanks Mike, I'll have to see how that works! In the worst case I may have to remove the roslaunch check from the system...quite a bummer.

closed 2017-12-20 by bmagyar